Meliox/PVE-mods #1: Sensors -j output

evoandroidevo, 2023-06-12:
Can you add an output of `sensors -j` that currently works with this so i can change it to work with my output from a different cpu

Meliox, 2023-06-14:
`root@proxmox:~# sensors -j
{
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 58.000,
         "temp1_max": 105.000,
         "temp1_crit": 105.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 53.000,
         "temp2_max": 105.000,
         "temp2_crit": 105.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 53.000,
         "temp3_max": 105.000,
         "temp3_crit": 105.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 53.000,
         "temp4_max": 105.000,
         "temp4_crit": 105.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 53.000,
         "temp5_max": 105.000,
         "temp5_crit": 105.000,
         "temp5_crit_alarm": 0.000
      }
   },
   "acpitz-acpi-0":{
      "Adapter": "ACPI interface",
      "temp1":{
         "temp1_input": 73.000,
         "temp1_crit": 119.000
      }
   },
   "nvme-pci-0100":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 65.850,
         "temp1_max": 82.850,
         "temp1_min": -5.150,
         "temp1_crit": 87.850,
         "temp1_alarm": 0.000
      }
   }
}`

Meliox, 2023-06-27:
Hi all,

You can try the new version just published. It will display of temperatures for all CPU cores, NVMe drives and HDD/SSDs dynamically.

evoandroidevo, 2023-06-27:
> Hi all,
> 
> You can try the new version just published. It will display of temperatures for all CPU cores, NVMe drives and HDD/SSDs dynamically.

[image]
amd cpu doesnt auto work with it heres my code that i used from the old version 

``` sed -i "/^Ext.define('PVE.node.StatusView',/ {
      :a;
      /items:/!{N;ba;}
      :b;
      /swap.*},/!{N;bb;}
      a\
      \\
        {\n\
            itemId: 'thermal',\n\
            colspan: 1,\n\
            printBar: false,\n\
            title: gettext('CPU Thermal State'),\n\
            iconCls: 'fa fa-fw fa-thermometer-half',\n\
            textField: 'thermalstate',\n\
            renderer: function(value){\n\
              let objValue = JSON.parse(value);\n\
              let core0 = objValue[\"k10temp-pci-00c3\"][\"Tctl\"][\"temp1_input\"];\n\
              let core1 = objValue[\"k10temp-pci-00c3\"][\"Tccd1\"][\"temp3_input\"];\n\
              let core2 = objValue[\"k10temp-pci-00c3\"][\"Tccd2\"][\"temp4_input\"];\n\
              return \`CPU Temp: \$\{core0\} C | CCD1 Temp: \$\{core1\} C | CCD2 Temp: \$\{core2\} C\`\n\
            }\n\
        },\n\
        {\n\
            itemId: 'thermal2',\n\
            colspan: 1,\n\
            printBar: false,\n\
            title: gettext('NVME Thermal State'),\n\
            iconCls: 'fa fa-fw fa-thermometer-half',\n\
            textField: 'thermalstate2',\n\
            renderer: function(value){\n\
              let objValue = JSON.parse(value);\n\
              let temp0 = objValue[\"nvme-pci-0100\"][\"Composite\"][\"temp1_input\"];\n\
              return \`NVME: \$\{temp0\} C\`\n\
            }\n\
        },
    }"```

Meliox, 2023-06-28:
Hmm. The naming for amd cpus is totally different.
Can you share the output of sensors -j.

evoandroidevo, 2023-06-28:
> […](#)

``` json
{
   "nvme-pci-0100":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 36.850,
         "temp1_max": 81.850,
         "temp1_min": -273.150,
         "temp1_crit": 84.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 36.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      },
      "Sensor 2":{
         "temp3_input": 41.850,
         "temp3_max": 65261.850,
         "temp3_min": -273.150
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 51.625
      },
      "Tccd1":{
         "temp3_input": 48.750
      },
      "Tccd2":{
         "temp4_input": 45.000
      }
   }
}
```

Meliox, 2023-06-28:
Hi,
Thanks for sharing. I tried to make a workaround for this by letting the user input the cpu configuration as I cannot guess it.

Can I ask you to try the above referred version?

commit b642d4c1 (2023-07-08): Ask user for input for CPU configuration (#4)

* More flexible CPU detection. Ask user to input correct parameters. Examples are presented to guide the user
* Amd temps are by default limited to "Tccd" whereas Intel is "Cores"
* Rework indexing of all temperature items. CPU temps will be presented in the order presented from lm-sensors
* Dynamically add temps and only show temps that are available (CPU, HDD, SSD, NVMe)
* Change default backup location to same location as the script
* Reworked script output messages
* Many code quality improvements
* Visual improvements temp display in proxmox  [... cut]
closed 2023-07-08 by Meliox