Meliox/PVE-mods #3: Proxmox 8?

sampl3x, 2023-06-25:
Does this script work on the new Proxmox version 8?

sampl3x, 2023-06-28:
Thank you for the update, it works!

Possible to make an option to not show HDD/SSD Thermal State when there is not HDD/SSD used? I have only nvme and give me a HDD/SSD Thermal State in de Summary.

Meliox, 2023-06-28:
Hi,
Please try the above referred version.

sampl3x, 2023-06-29:
Sorry,I don't know how to get that version.

Meliox, 2023-06-29:
https://raw.githubusercontent.com/Meliox/PVE-mods/d63e83c62979aa3c22035f39359066a62df376f3/pve-mod-gui-temp.sh

sampl3x, 2023-06-29:
Thank you, i tested it and it only show the NVME but i dont know what to fill in on the questions, now no cpu info is not shown.
Enter the cpu adapter name: ???
Enter the cpu item prefix: ???

Meliox, 2023-06-29:
What is shown when you type sensors - j?

Meliox, 2023-07-08:
Resolved in b642d4c

closed 2023-07-08 by Meliox