Meliox/PVE-mods #5: Core 4: undefined°C | Almost working

mrkoopa1, 2023-07-03:
I have one of those CWWK aliexpress boxes with the intel  i7-1265U so 2 performance cores, and 8 efficiency cores. The script did run but something not quite right. Attached is the sensors -j output and what the dashboard is showing. I feel like I am just missing something in the variables. Any ideas?

```root@pvehome:~# sensors -j
{
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 50.000,
         "temp1_max": 100.000,
         "temp1_crit": 100.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 50.000,
         "temp2_max": 100.000,
         "temp2_crit": 100.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 4":{
         "temp3_input": 44.000,
         "temp3_max": 100.000,
         "temp3_crit": 100.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 8":{
         "temp4_input": 44.000,
         "temp4_max": 100.000,
         "temp4_crit": 100.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 9":{
         "temp5_input": 44.000,
         "temp5_max": 100.000,
         "temp5_crit": 100.000,
         "temp5_crit_alarm": 0.000
      },
      "Core 10":{
         "temp6_input": 44.000,
         "temp6_max": 100.000,
         "temp6_crit": 100.000,
         "temp6_crit_alarm": 0.000
      },
      "Core 11":{
         "temp7_input": 44.000,
         "temp7_max": 100.000,
         "temp7_crit": 100.000,
         "temp7_crit_alarm": 0.000
      },
      "Core 12":{
         "temp8_input": 43.000,
         "temp8_max": 100.000,
         "temp8_crit": 100.000,
         "temp8_crit_alarm": 0.000
      },
      "Core 13":{
         "temp9_input": 43.000,
         "temp9_max": 100.000,
         "temp9_crit": 100.000,
         "temp9_crit_alarm": 0.000
      },
      "Core 14":{
         "temp10_input": 43.000,
         "temp10_max": 100.000,
         "temp10_crit": 100.000,
         "temp10_crit_alarm": 0.000
      },
      "Core 15":{
         "temp11_input": 43.000,
         "temp11_max": 100.000,
         "temp11_crit": 100.000,
         "temp11_crit_alarm": 0.000
      }
   },
   "acpitz-acpi-0":{
      "Adapter": "ACPI interface",
      "temp1":{
         "temp1_input": 27.800,
         "temp1_crit": 105.000
      }
   },
   "nvme-pci-0100":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 59.850,
         "temp1_max": 81.850,
         "temp1_min": -273.150,
         "temp1_crit": 84.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 59.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      },
      "Sensor 2":{
         "temp3_input": 68.850,
         "temp3_max": 65261.850,
         "temp3_min": -273.150
      }
   }
}
```
[image]

Meliox, 2023-07-03:
Hi, 
Thanks for the report. We are working on a new version. Can you try this one and report back? https://github.com/Meliox/PVE-mods/blob/6ad9633c7e1d4488f5efbfe5163be314420e1b2d/pve-mod-gui-temp.sh

mrkoopa1, 2023-07-03:
Thanks for the info. Looks better, seeing everything now. Not sure if it will be an issue for anyone else but the core IDs don't match sensors -j output. For whatever reason the 2 P-cores are 0 and 4 then the E-cores start at 8-15 Dashboard shows 0-9

[image]

Meliox, 2023-07-03:
Currently an incrementing number is just added. It could be considered to extract from the Core.

From your sensor -j output, P and E cores cannot be identified. 

I also note that only one NVME drive is showing up, which requires a fix.

Meliox, 2023-07-08:
Ressolved in b642d4c

closed 2023-07-08 by Meliox