Meliox/PVE-mods #7: Subscription nag screen removal

eremem, 2023-07-09:
Since a paid subscription is unlikely to be an option for private use (at least in my case), watching a nag screen on a regular basis started to annoy me more and more. So I implemented a script that blocks the popup dialog.
Such a modification is of course somehow controversial, so I'm curious if there should be a place for such a function in the project.

LordMyschkin, 2023-07-17:
As you say, from the very beginning, this nag screen does not motivate private people to pay for a subscription. i'd be very interested in that script.

Meliox, 2023-07-17:
I really like to idea of collecting everything that can help our installation. However, I am not sure that is against the terms (Proxmox Virtual Environment's source code is published under the free software license GNU AGPL, v3 and thus is freely available for download, use and share). Do you know?

Currently, such a script can be found in other repos here on github.

eremem, 2023-07-17:
Since the code is open source, it is allowed to be modified. We provide tools which change the code, so we don't have to share the original code with the adjustments. From my point of view removal of the nag screen could be considered somewhat unethical, however this seems also far-fetched to me. The mods can be applied selectively, who still wants to see the nag screen, can choose to do so.

eremem, 2023-07-18:
I pushed the script into the main repository. I also added a second one, which allows to install/uninstall all of the mods - the list of the scripts to call is configurable and can be easily extended in the future.

closed 2023-07-19 by eremem