Meliox/PVE-mods #8: Single temperature for high-core CPU's

to4ko, 2023-07-17:
Hi! 

Is it possible to modify script to get single number (Package id 0) readings on summary screen instead of 14?)

[image]

Meliox, 2023-07-17:
Hello,
Should work if you change this line: https://github.com/Meliox/PVE-mods/blob/main/pve-mod-gui-temp.sh#L115C6-L115C21 to "Package id " instead.

to4ko, 2023-07-17:
> Hello, Should work if you change this line: https://github.com/Meliox/PVE-mods/blob/main/pve-mod-gui-temp.sh#L115C6-L115C21 to "Package id " instead.

i'm doing something wrong?

[image]

Meliox, 2023-07-17:
I mean change it from: CPU_ITEM_PREFIX="Core " to   CPU_ITEM_PREFIX="Package
id ". However, the line indicate something is wrong in or around line 15.
https://github.com/Meliox/PVE-mods/blob/main/pve-mod-gui-temp.sh#L15

to4ko, 2023-07-17:
@Meliox small one - how this empy space can be removed?

[image]

Meliox, 2023-07-17:
This is not caused by the mod, hence, I would ask, or fill a bug, to the proxmox team.

Btw. If you want to change "Core 0" to "Package 0", then simply change this line: https://github.com/Meliox/PVE-mods/blob/main/pve-mod-gui-temp.sh#L116

Non-moded:
[image]

to4ko, 2023-07-17:
it;s strange - here is mine non-moded
[image]

Meliox, 2023-07-17:
Can you share your non-modded file for comparison?

to4ko, 2023-07-17:
attached
[pvemanagerlib_and_Nodes_pm.zip](https://github.com/Meliox/PVE-mods/files/12073330/pvemanagerlib_and_Nodes_pm.zip)

Meliox, 2023-07-17:
Thanks.
Ok, mine had some leftovers:
[image]

So, I guess we can conclude that something in this section makes in unwanted behavior. 

https://github.com/Meliox/PVE-mods/blob/main/pve-mod-gui-temp.sh#L190

eremem, 2023-07-19:
@Meliox I doesn't really look like our bug. Regardless of the modified `height`, `bodyPadding` and the other properties, the actual height of the box must grow anyway due to the added temperature entries. The layout manager of ExtJS (the JS framework under the hood) doesn't seem to be able to vertically align the boxes if their heights differ.

carlox97, 2023-11-02:
I edited [Line 15 and 16](https://github.com/Meliox/PVE-mods/blob/main/pve-mod-gui-temp.sh#L115) to
`CPU_ITEM_PREFIX="Package id "`
`CPU_TEMP_CAPTION="CPU Package"`

[Line 233](https://github.com/Meliox/PVE-mods/blob/main/pve-mod-gui-temp.sh#L233) to
``tempStr = \`\${cpuTempCaption}:&nbsp;\${temp}&deg;C\`;\n\``

to get "CPU Package: 52°C"

and [Line 325](https://github.com/Meliox/PVE-mods/blob/main/pve-mod-gui-temp.sh#L325) to
``temps.push(\`Samsung&nbsp;\EVO&nbsp;\960:&nbsp;\${temp}&deg;C\`);\n\``

to get "Samsung EVO 960: 27.85°C"

[image]

@Meliox do you think it's fine or it might cause problems? I've had none so far.

Meliox, 2023-11-02:
Looks great. It will not cause you any problems. You will need to modify the script again if this one is updated.

Meliox, 2024-06-16:
Implemented in #54 . Closing this

closed 2024-06-16 by Meliox