Meliox/PVE-mods #11: my CPU doesn't display temperature

Sh4Dz, 2023-08-12:
Hi,

After running the script, everything went well, but my CPU doesn't display its temperature. However, it's listed as a recognized CPU: "k10temp-pci-xxx".
[image]

Here is my sensors -j :

   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 51.375
      }

Meliox, 2023-08-12:
Hi,
For some reason the Tccd is not available for your cpu. Instead we have tctl. To make it work seamlessly we need to update the indetification logic, for the easiest and quickest thing for you to do to get it working is to uninstall and change the qouted prefix here https://github.com/Meliox/PVE-mods/blob/f4e7aa44559e5221139f9ab54f3e887118fe16b1/pve-mod-gui-temp.sh#L121 to "Tctl" and then install again.
I will have a look at the implementation when I am back from vacation. 

Also, thanks for providing GPU info, that information can be added as well :) just need to be sure. Is it from a dedicated GPU or an inbuit graphics chip on the Cpu?

Sh4Dz, 2023-08-13:
Hi,
Thank you for your response.
[image]
For the gpu it's an inbuilt graphic chip on the cpu.

Damien0505, 2024-01-02:
Hi,
Hoping you may be able to assist.  I have a Dell Precision T5600 with dual  Xeon E5-2690's.
Unfortunately nothing is populating after installing the script.  I have provided the output of `sensors -j` for your reference:

> root@pve3:~# sensors -j
> 
> {
>    "nouveau-pci-0400":{
>       "Adapter": "PCI adapter",
>       "GPU core":{
>          "in0_input": 0.920,
>          "in0_min": 0.920,
>          "in0_max": 1.000
>       },
>       "fan1":{
>          "fan1_input": 2460.000
>       },
>       "temp1":{
>          "temp1_input": 45.000,
>          "temp1_max": 95.000,
>          "temp1_max_hyst": 3.000,
>          "temp1_crit": 105.000,
>          "temp1_crit_hyst": 5.000,
>          "temp1_emergency": 135.000,
>          "temp1_emergency_hyst": 5.000
>       }
>    },
>    "coretemp-isa-0001":{
>       "Adapter": "ISA adapter",
>       "Package id 1":{
>          "temp1_input": 48.000,
>          "temp1_max": 85.000,
>          "temp1_crit": 95.000,
>          "temp1_crit_alarm": 0.000
>       },
>       "Core 0":{
>          "temp2_input": 48.000,
>          "temp2_max": 85.000,
>          "temp2_crit": 95.000,
>          "temp2_crit_alarm": 0.000
>       },
>       "Core 1":{
>          "temp3_input": 41.000,
>          "temp3_max": 85.000,
>          "temp3_crit": 95.000,
>          "temp3_crit_alarm": 0.000
>       },
>       "Core 2":{
>          "temp4_input": 46.000,
>          "temp4_max": 85.000,
>          "temp4_crit": 95.000,
>          "temp4_crit_alarm": 0.000
>       },
>       "Core 3":{
>          "temp5_input": 43.000,
>          "temp5_max": 85.000,
>          "temp5_crit": 95.000,
>          "temp5_crit_alarm": 0.000
>       },
>       "Core 4":{
>          "temp6_input": 47.000,
>          "temp6_max": 85.000,
>          "temp6_crit": 95.000,
>          "temp6_crit_alarm": 0.000
>       },
>       "Core 5":{
>          "temp7_input": 43.000,
>          "temp7_max": 85.000,
>          "temp7_crit": 95.000,
>          "temp7_crit_alarm": 0.000
>       },
>       "Core 6":{
>          "temp8_input": 46.000,
>          "temp8_max": 85.000,
>          "temp8_crit": 95.000,
>          "temp8_crit_alarm": 0.000
>       },
>       "Core 7":{
>          "temp9_input": 47.000,
>          "temp9_max": 85.000,
>          "temp9_crit": 95.000,
>          "temp9_crit_alarm": 0.000
>       }
>    },
>    "dell_smm-isa-0000":{
>       "Adapter": "ISA adapter",
>       "Processor Fan":{
>          "fan1_input": 974.000,
>          "fan1_min": 0.000,
>          "fan1_max": 5970.000
>       },
>       "Processor Fan":{
>          "fan2_input": 2984.000,
>          "fan2_min": 0.000,
>          "fan2_max": 5970.000
>       },
>       "Other Fan":{
>          "fan3_input": 1359.000,
>          "fan3_min": 0.000,
>          "fan3_max": 3914.000
>       },
>       "CPU":{
>          "temp1_input": 51.000
>       },
>       "CPU":{
>          "temp2_input": 53.000
>       },
>       "SODIMM":{
>          "temp3_input": 42.000
>       },
>       "SODIMM":{
>          "temp4_input": 41.000
>       },
>       "Ambient":{
>          "temp5_input": 28.000
>       },
>       "Ambient":{
>          "temp6_input": 29.000
>       },
>       "CPU":{
>          "temp7_input": 33.000
>       },
>       "CPU":{
>          "temp8_input": 43.000
>       },
>       "SODIMM":{
>          "temp9_input": 37.000
>       },
>       "SODIMM":{
>          "temp10_input": 40.000
>       }
>    },
>    "coretemp-isa-0000":{
>       "Adapter": "ISA adapter",
>       "Package id 0":{
>          "temp1_input": 48.000,
>          "temp1_max": 85.000,
>          "temp1_crit": 95.000,
>          "temp1_crit_alarm": 0.000
>       },
>       "Core 0":{
>          "temp2_input": 46.000,
>          "temp2_max": 85.000,
>          "temp2_crit": 95.000,
>          "temp2_crit_alarm": 0.000
>       },
>       "Core 1":{
>          "temp3_input": 47.000,
>          "temp3_max": 85.000,
>          "temp3_crit": 95.000,
>          "temp3_crit_alarm": 0.000
>       },
>       "Core 2":{
>          "temp4_input": 47.000,
>          "temp4_max": 85.000,
>          "temp4_crit": 95.000,
>          "temp4_crit_alarm": 0.000
>       },
>       "Core 3":{
>          "temp5_input": 48.000,
>          "temp5_max": 85.000,
>          "temp5_crit": 95.000,
>          "temp5_crit_alarm": 0.000
>       },
>       "Core 4":{
>          "temp6_input": 47.000,
>          "temp6_max": 85.000,
>          "temp6_crit": 95.000,
>          "temp6_crit_alarm": 0.000
>       },
>       "Core 5":{
>          "temp7_input": 45.000,
>          "temp7_max": 85.000,
>          "temp7_crit": 95.000,
>          "temp7_crit_alarm": 0.000
>       },
>       "Core 6":{
>          "temp8_input": 45.000,
>          "temp8_max": 85.000,
>          "temp8_crit": 95.000,
>          "temp8_crit_alarm": 0.000
>       },
>       "Core 7":{
>          "temp9_input": 45.000,
>          "temp9_max": 85.000,
>          "temp9_crit": 95.000,
>          "temp9_crit_alarm": 0.000
>       }
>    }
> }
> root@pve3:~#
> 

Any advice would be greatly appreciated!
Regards,
Damien

Meliox, 2024-01-02:
Hi Damien,
Sensor -j output looks good and script should work out-of-the-box. Did you get any errors during installation and/or refresh the webui of proxmox (CTRL+F5)? If, yes, what do you see there.

Damien0505, 2024-01-05:
No errors at all:

> root@pve3:~# ./pve-mod-gui-temp.sh install
> 
> Installing the Proxmox VE temperatures display mod...
> 
> Preparing mod installation...
> 
> Detecting support for HDD/SDD temperature sensors...
> 
> Detecting support for NVMe temperature sensors...
> [warning] No NVMe temperature sensors found.
> 
> Detecting support for CPU temperature sensors...
> Detected sensor:
> coretemp-isa-0001
> coretemp-isa-0000
> 
> Backup of "/usr/share/perl5/PVE/API2/Nodes.pm" saved to "/root/backup/Nodes.pm.2024-01-06_03-15-29".
> Added thermalstate to /usr/share/perl5/PVE/API2/Nodes.pm.
> Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/root/backup/pvemanagerlib.js.2024-01-06_03-15-29".
> Expanded space in "/usr/share/pve-manager/js/pvemanagerlib.js".
> New temperature display items added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".
> 
> Restarting PVE proxy...
> Installation completed
> 
> root@pve3:~#
> 

Errors are present in Chrome dev console:

[image]

[image]

Kind regards,
Damien

eremem, 2024-05-11:
Please check the latest version. The attached `sensors -j` output works with it w/o any errors.

closed 2024-05-11 by eremem