Meliox/PVE-mods #13: Not seeing temperature on summary - Sed Expression

Alternating, 2023-08-25:
sed: -e expression #1, char 382: unknown command: `k' 

`root@proxmox:~# ./pve-mod-gui-temp.sh install

Installing the Proxmox VE temperatures display mod...

Preparing mod installation...
ERROR: Can't get value of subfeature temp1_input: Can't read

Detecting support for HDD/SDD temperature sensors...

Detecting support for NVMe temperature sensors...
[warning] No NVMe temperature sensors found.

Detecting support for CPU temperature sensors...
Detected sensor:
k10temp-pci-00c3
k10temp-pci-00cb

Backup of "/usr/share/perl5/PVE/API2/Nodes.pm" saved to "/root/backup/Nodes.pm.2023-08-25_12-34-14".
Added thermalstate to /usr/share/perl5/PVE/API2/Nodes.pm.
Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/root/backup/pvemanagerlib.js.2023-08-25_12-34-14".
Expanded space in "/usr/share/pve-manager/js/pvemanagerlib.js".
**_sed: -e expression #1, char 382: unknown command: `k'_**
New temperature display items added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".

Restarting PVE proxy...
Installation completed
`

Meliox, 2023-08-25:
Please execute the script with bash script.sh. I think that will solve your issue. If not please provide output of bash - X script.sh and an output of sensor -j.

Thank you

Alternating, 2023-08-25:
**pve-mod-gui-temp.sh**

root@proxmox:~# bash pve-mod-gui-temp.sh install

Installing the Proxmox VE temperatures display mod...

Preparing mod installation...
ERROR: Can't get value of subfeature temp1_input: Can't read

Detecting support for HDD/SDD temperature sensors...

Detecting support for NVMe temperature sensors...
[warning] No NVMe temperature sensors found.

Detecting support for CPU temperature sensors...
Detected sensor:
k10temp-pci-00c3
k10temp-pci-00cb

[warning] Thermalstate already added to "/usr/share/perl5/PVE/API2/Nodes.pm".
Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/root/backup/pvemanagerlib.js.2023-08-25_15-30-56".
Expanded space in "/usr/share/pve-manager/js/pvemanagerlib.js".
sed: -e expression #1, char 382: unknown command: `k'
New temperature display items added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".

Restarting PVE proxy...
Installation completed

**SENSORS -j**

root@proxmox:~# sensors -j
{
   "radeon-pci-4100":{
      "Adapter": "PCI adapter",
      "temp1":{
         "temp1_input": 37.000,
         "temp1_crit": 120.000,
         "temp1_crit_hyst": 90.000
      }
   },
   "asus_wmi_sensors-virtual-0":{
      "Adapter": "Virtual device",
      "CPU Core Voltage":{
         "in0_input": 1.373
      },
      "CPU SOC Voltage":{
         "in1_input": 0.916
      },
      "DRAM AB Voltage":{
         "in2_input": 1.177
      },
      "DRAM CD Voltage":{
         "in3_input": 1.199
      },
      "1.8V PLL Voltage":{
         "in4_input": 1.766
      },
      "+12V Voltage":{
         "in5_input": 11.772
      },
      "+5V Voltage":{
         "in6_input": 4.878
      },
      "3VSB Voltage":{
         "in7_input": 3.270
      },
      "VBAT Voltage":{
         "in8_input": 3.030
      },
      "AVCC3 Voltage":{
         "in9_input": 3.314
      },
      "SB 1.05V Voltage":{
         "in10_input": 1.041
      },
      "CPU Core Voltage":{
         "in11_input": 1.355
      },
      "CPU SOC Voltage":{
         "in12_input": 0.900
      },
      "DRAM AB Voltage":{
         "in13_input": 1.200
      },
      "DRAM CD Voltage":{
         "in14_input": 1.200
      },
      "CPU Fan":{
         "fan1_input": 1227.000
      },
      "Chassis Fan 1":{
         "fan2_input": 0.000
      },
      "Chassis Fan 2":{
         "fan3_input": 0.000
      },
      "HAMP Fan":{
         "fan4_input": 0.000
      },
      "Water Pump 1":{
         "fan5_input": 0.000
      },
      "CPU OPT":{
         "fan6_input": 0.000
      },
      "Water Flow":{
         "fan7_input": 0.000
      },
      "Waterblock Flow":{
         "fan8_input": 0.000
      },
      "EXT Fan 1":{
         "fan9_input": 0.000
      },
      "EXT Fan 2":{
         "fan10_input": 0.000
      },
      "EXT Fan 3":{
         "fan11_input": 0.000
      },
      "Cover Fan":{
         "fan12_input": 0.000
      },
      "CPU Temperature":{
         "temp1_input": 24.000
      },
      "CPU Socket Temperature":{
         "temp2_input": 44.000
      },
      "Motherboard Temperature":{
         "temp3_input": 36.000
      },
      "Chipset Temperature":{
         "temp4_input": 57.000
      },
      "Tsensor 1 Temperature":{
         "temp5_input": 216.000
      },
      "CPU VRM Temperature":{
         "temp6_input": 49.000
      },
      "Water In":{
         "temp7_input": 216.000
      },
      "Water Out":{
         "temp8_input": 216.000
      },
      "Waterblock In":{
         "temp9_input": 216.000
      },
      "Waterblock Out":{
         "temp10_input": 216.000
      },
      "EXT Tsensor 1":{
         "temp11_input": 0.000
      },
      "EXT Tsensor 2":{
         "temp12_input": 0.000
      },
      "EXT Tsensor 3":{
         "temp13_input": 0.000
      },
      "Tsensor 2 Temperature":{
         "temp14_input": 216.000
      },
      "DIMM.2 Tsensor 1 Temperature":{
         "temp15_input": 216.000
      },
      "DIMM.2 Tsensor 2 Temperature":{
         "temp16_input": 216.000
      },
      "CPU VRM Output Current":{
         "curr1_input": 47.000
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 51.000
      },
      "Tdie":{
         "temp2_input": 24.000
      },
      "Tccd1":{
         "temp3_input": 51.500
      }
   },
   "ath10k_hwmon-pci-0300":{
      "Adapter": "PCI adapter",
      "temp1":{
ERROR: Can't get value of subfeature temp1_input: Can't read

      }
   },
   "k10temp-pci-00cb":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 51.000
      },
      "Tdie":{
         "temp2_input": 24.000
      }
   }
}

eremem, 2024-05-11:
The current version contains a better error detection and should be able to handle errors reported by `sensors` command.
I'm closing this issue. If the problem persist, please reopen this issue or open a new one.

closed 2024-05-11 by eremem