Meliox/PVE-mods #14: [Feature Request] - Adding GPU and RAM temperature

wes1993, 2023-09-13:
Hello, 
tanks a lot for you work!!

Could you think is possible to add support for GPU and RAM?

Best Regards
Stefano

Meliox, 2023-09-14:
Hey,

Thanks.

If the info is available. Can you please share your sensor - j output?

wes1993, 2023-09-14:
Hello @Meliox , thanks a lot for your reply 

``
{
   "drivetemp-scsi-5-0":{
      "Adapter": "SCSI adapter",
      "temp1":{
         "temp1_input": 40.000
      }
   },
   "drivetemp-scsi-3-0":{
      "Adapter": "SCSI adapter",
      "temp1":{
         "temp1_input": 41.000,
         "temp1_max": 0.000,
         "temp1_min": 0.000,
         "temp1_crit": 0.000,
         "temp1_lcrit": 0.000,
         "temp1_lowest": 41.000,
         "temp1_highest": 42.000
      }
   },
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 55.000,
         "temp1_max": 86.000,
         "temp1_crit": 100.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 54.000,
         "temp2_max": 86.000,
         "temp2_crit": 100.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 52.000,
         "temp3_max": 86.000,
         "temp3_crit": 100.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 55.000,
         "temp4_max": 86.000,
         "temp4_crit": 100.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 52.000,
         "temp5_max": 86.000,
         "temp5_crit": 100.000,
         "temp5_crit_alarm": 0.000
      }
   },
   "drivetemp-scsi-4-0":{
      "Adapter": "SCSI adapter",
      "temp1":{
         "temp1_input": 42.000,
         "temp1_max": 0.000,
         "temp1_min": 0.000,
         "temp1_crit": 0.000,
         "temp1_lcrit": 0.000,
         "temp1_lowest": 42.000,
         "temp1_highest": 43.000
      }
   }
``

Meliox, 2023-09-15:
It does not seem that GPU (or memory) temperature are present in the output.

According the lm-sensors instructions you can try the following:

To use sensors, follow these steps:

Install the lm-sensors package using your Linux distribution’s package manager.
Run the sensors-detect command as root to detect and configure sensors on your system.
Run the sensors command to display the sensor readings, including GPU temperature.

wes1993, 2023-09-15:
Hello @Meliox ,
Is possible that this is because is an Nvidia GPU?

Meliox, 2023-09-18:
I read that it should be possible by loading the correct module, but many recommend using
# nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader,nounits
Instead. Nvidia provide its own tools. Is it already installed as part of the Nvidia driver package? Can you install and run above?

Dimm should be possible with some configuration. https://wiki.archlinux.org/title/lm_sensors. I don't think it worth the extra effort. Configuration would have to be done by the user.

wes1993, 2023-09-23:
Thanks again for your reply @Meliox,
Unfortunately i can't install the driver in the Host because i Need the GPU for the Guest Machines

Meliox, 2023-09-23:
If you find another way. Please reach out.

wes1993, 2024-07-23:
Hello Meliox, finally I have installed the Nvidia drivers in my host and now the command below works:
```
nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader,nounits
```

The output is the temperature:
```
root@pve:~# nvidia-smi --query-gpu=temperature.gpu --format=csv,noheader,nounits
47

```

Could you please add this to your script?

Best regards
Stefano

Meliox, 2024-07-23:
Thanks for sharing. Yes, I can look into that when I have time.

wes1993, 2024-07-23:
For the fan speed the command is:
```
nvidia-smi --query-gpu=fan.speed --format=csv,noheader,nounits
```
Here the command output
```
root@pve:~# nvidia-smi --query-gpu=fan.speed --format=csv,noheader,nounits
39
```

gluca, 2025-01-02:
here my sensors -j output. Ram temperature is in the SODIMM section
root@t7910:~# sensors -j
{
   "coretemp-isa-0001":{
      "Adapter": "ISA adapter",
      "Package id 1":{
         "temp1_input": 45.000,
         "temp1_max": 75.000,
         "temp1_crit": 85.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 37.000,
         "temp2_max": 75.000,
         "temp2_crit": 85.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 38.000,
         "temp3_max": 75.000,
         "temp3_crit": 85.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 40.000,
         "temp4_max": 75.000,
         "temp4_crit": 85.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 39.000,
         "temp5_max": 75.000,
         "temp5_crit": 85.000,
         "temp5_crit_alarm": 0.000
      },
      "Core 4":{
         "temp6_input": 39.000,
         "temp6_max": 75.000,
         "temp6_crit": 85.000,
         "temp6_crit_alarm": 0.000
      },
      "Core 5":{
         "temp7_input": 38.000,
         "temp7_max": 75.000,
         "temp7_crit": 85.000,
         "temp7_crit_alarm": 0.000
      }
   },
   "dell_smm-isa-0000":{
      "Adapter": "ISA adapter",
      "Processor Fan":{
         "fan1_input": 995.000,
         "fan1_min": 0.000,
         "fan1_max": 5755.000
      },
      "Processor Fan":{
         "fan2_input": 994.000,
         "fan2_min": 0.000,
         "fan2_max": 5755.000
      },
      "Other Fan":{
         "fan3_input": 600.000,
         "fan3_min": 0.000,
         "fan3_max": 5000.000
      },
      "CPU":{
         "temp1_input": 40.000
      },
      "CPU":{
         "temp2_input": 45.000
      },
      "SODIMM":{
         "temp3_input": 25.000
      },
      "SODIMM":{
         "temp4_input": 33.000
      },
      "SODIMM":{
         "temp5_input": 33.000
      },
      "SODIMM":{
         "temp6_input": 24.000
      }
   },
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 40.000,
         "temp1_max": 75.000,
         "temp1_crit": 85.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 36.000,
         "temp2_max": 75.000,
         "temp2_crit": 85.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 35.000,
         "temp3_max": 75.000,
         "temp3_crit": 85.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 35.000,
         "temp4_max": 75.000,
         "temp4_crit": 85.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 36.000,
         "temp5_max": 75.000,
         "temp5_crit": 85.000,
         "temp5_crit_alarm": 0.000
      },
      "Core 4":{
         "temp6_input": 36.000,
         "temp6_max": 75.000,
         "temp6_crit": 85.000,
         "temp6_crit_alarm": 0.000
      },
      "Core 5":{
         "temp7_input": 35.000,
         "temp7_max": 75.000,
         "temp7_crit": 85.000,
         "temp7_crit_alarm": 0.000
      }
   }
}

Meliox, 2025-01-02:
@gluca I think you're making a different feature - to show ram temps, right? Please open a separate issue. Thanks.

Meliox, 2025-01-06:
Ram temperatures will be added in #75. However, they must be present under SODIMM. @wes1993: I do not see any ram temps in your sensor -j output, so you need to resolve this first.

Meliox, 2025-01-06:
I am changing this issue to only cover NVIDIA GPU suppport for the host as ram will be implemented.