Meliox/PVE-mods #16: Scripts installed but no results.

nightfever77, 2023-10-16:
Installed without error, pveproxy reloaded.
Repeated installation is telling:

`...
Detecting support for NVMe temperature sensors...
Detected sensors:
nvme-pci-3a00

Detecting support for CPU temperature sensors...
Detected sensor:
coretemp-isa-0000

[warning] Thermalstate already added to "/usr/share/perl5/PVE/API2/Nodes.pm".
[warning] New temperature display items already added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".`

**But still no temp sensors in summary !** 

Same with pve-mod-nag-screen.sh, still getting nag screens.

Is there anything elso to do?

Ryder1021, 2024-01-09:
I received this error when running the installer
.......
Backup of "/usr/share/perl5/PVE/API2/Nodes.pm" saved to "/root/backup/Nodes.pm.2024-01-08_22-49-16".
Added thermalstate to /usr/share/perl5/PVE/API2/Nodes.pm.
Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/root/backup/pvemanagerlib.js.2024-01-08_22-49-16".
Expanded space in "/usr/share/pve-manager/js/pvemanagerlib.js".
sed: -e expression #1, char 382: unknown command: `k'

When looking at the pvemanagerlib.js file it doesn't appear there have been any additions to it, possibly due to the error?

Hope this is the right place for this, sorry if it isn't and thanks for any answer.

Meliox, 2024-01-09:
How did you execute the installer?

Ryder1021, 2024-01-09:
Bash pve-mod-gui-temp.sh install after using wget to download it.

Ryder1021, 2024-01-13:
These are my 2 CPU die output strings from "sensors -j" (I have Ryzen Threadripper)

  "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 58.625
      },
      "Tdie":{
         "temp2_input": 31.625
      },
      "Tccd1":{
         "temp3_input": 58.500
      

 "k10temp-pci-00cb":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 58.500
      },
      "Tdie":{
         "temp2_input": 31.500

I am not the best at reading code, but does the script fail because of "Tccd1" under the "k10temp-pci-00c3"?
It seems like the script isn't getting the right info from my CPU and thus generating that SED error when trying to patch the pvemanagerlib.js
Neither of the detected "k10temp" items have the right prefix to populate the script. At least from my very untrained eye.

The script worked just fine on my other Proxmox server that runs a regular Ryzen 1600x

Thank you!

closed 2024-04-10 by Meliox