Meliox/PVE-mods #17: system without lm-sensors installed sits on a loading screen

MasterCATZ, 2023-11-04:
if I log into a Proxmox GUI with any server that had lm-sensors installed temperatures will show ,

but for systems that do not have lm-sensors installed the entire page becomes frozen 

any way to put a check in place so it will skip trying to load data that does not exist ? 

[image]
[image]
[image]
[image]

Meliox, 2023-11-05:
Hi,

There may be other issues as well, as a node setup was not considered. The easiest approach may be to install lm-sensors.

Anyhow, as I don't have such a setup available, I need you test test the approach to understand where it stops

In /usr/share/perl5/PVE/API2/Nodes.pm

Find the line:  $res->{thermalstate} = `sensors -j`;

replace `sensors -j` with ""

Refresh your browser - does the freezing still happen - all information should be empty.

eremem, 2024-05-11:
@Meliox I added an additional check for missing `sensors` output which should also fix the loading problem reported here.

Meliox, 2024-05-27:
Closing this as reported fixed. Feel free to reopen if it still is a problem.

closed 2024-05-27 by Meliox