Meliox/PVE-mods #19: Help Identifying temps pulled...

bearhntr, 2024-02-01:
Thanks for the script - and reading through your README - you had mentioned the DRIVETEMP KERNAL MODULE, no real instructions on making sure it was there - as I had installed lm-sensors but was not seeing anything that looked like HDD temps.  My PVE box has one NVME drive via a card in a PCIe slot and an SSD drive attached to SATA port on motherboard.

After doing the steps here:   
[image]

I am seeing this - but how do I know which is which for HDDs?   Also - what are these other temps?

[image]

Meliox, 2024-02-01:
To read HDD temperatures you need to use hwmon, which can be accessed using sensors when the drivetemp module is loaded.

`sudo apt install lm-sensors
sudo modprobe drivetemp
sensors`
To make drivetemp load automatically every boot, add it to /etc/modules as follows:

`echo drivetemp | sudo tee -a /etc/modules`

To identify drives you could do the following
`sudo smartctl -A /dev/sd(your drive letter) ` and grep temperature.

Isa that's your cpu.
Acpitz is the temperature sensor near/on your CPU socket. This sensor can be unreliable. 
Drivetemp-SCSI is HDD or SSD. 

I do not see NVME mentioned, but maybe the pci simply makes it look different.

bearhntr, 2024-02-01:
Thanks for the response - see my replies below.  

> To read HDD temperatures you need to use hwmon, which can be accessed using sensors when the drivetemp module is loaded.

  This does not install...Proxmox is basically DEBIAN at it's heart/core.  I tried to install with ```apt install hwmon``` and with ```apt-get install hwmon```   -- both fail.  

> `sudo apt install lm-sensors sudo modprobe drivetemp sensors` 
>  To make drivetemp load automatically every boot, add it to /etc/modules as follows:

I tried this - and got the following (does not appear that I need 'sudo':
[image]

 > `echo drivetemp | sudo tee -a /etc/modules`
> 
> To identify drives you could do the following `sudo smartctl -A /dev/sd(your drive letter) ` and grep temperature.
> 
> Isa that's your cpu. Acpitz is the temperature sensor near/on your CPU socket. This sensor can be unreliable. Drivetemp-SCSI is HDD or SSD.
> 
> I do not see NVME mentioned, but maybe the pci simply makes it look like a SSD/HDD

```sensors```  is telling me 2 HDDs

drivetemp-scsi-2-0
Adapter: SCSI adapter
temp1:        +22.0°C 

drivetemp-scsi-2-0
Adapter: SCSI adapter
temp1:        +22.0°C 

  Now to figure out how to get those to show properly in the Proxmox dashboard.

bearhntr, 2024-02-02:
I also have just tried to run the script - to see what would happen.  Apparently, all I get is flashing cursor until I CTRL+C to get out:

[image]

Meliox, 2024-02-02:
You only need to run 
`wget https://raw.githubusercontent.com/Meliox/PVE-mods/main/pve-mod-gui-temp.sh`

Notice that your command contains quite much more.

bearhntr, 2024-02-02:
> wget https://raw.githubusercontent.com/Meliox/PVE-mods/main/pve-mod-gui-temp.sh

I get this:   

[image]

...it looks to have saved a file (   ) in the root folder.  

The command that I was using - I have used numerous times to pull down a .sh file and RUN it with one command, does that not work with your .sh files?   How would I then run the .sh file which looks to be here now (I am sorry - if I seem 'daff', but I am not a Linux guru, and trying to learn as I go):

[image]

bearhntr, 2024-02-02:
OK - I think .... MAYBE ... I figured it out.  

[image]

I had already used the steps here to add a CPU temp.  Should I 'un-do' these to use your script?  
  [https://www.reddit.com/r/homelab/comments/rhq56e/displaying_cpu_temperature_in_proxmox_summery_in/]

Meliox, 2024-02-02:
Should be working now. Open the proxmox url and make force refresh, ctlr+f5 or clear the cache.

bearhntr, 2024-02-02:
> Should be working now. Open the proxmox url and make force refresh, ctlr+f5 or clear the cache.

I un-did all the changes I had manually done, and rebooted.  Dashboard was back to OOB state.  I then executed your .sh file, and low-and-behold...I got temps showing.  

[image]

❓  Is it possible to make the window where all the data is displayed - into the full width of the window, and the graphs below it?  Kinda like this (this is a created image using photoshop - to show desired view)  --  Maybe even add stuff or stretch things out?

[image]

Meliox, 2024-02-03:
Well anything is possible and I do agree your mockup does look better

bearhntr, 2024-02-03:
> Well anything is possible and I do agree your mockup does look better

I wish I understood more of the programming language used (whatever it is).  I can kinda read the 'code' and figure out what most of it is.... (see image) I believe this is the "start' of the information box that is displaying all the information.  Near the top it mentions 'minHeight:' - but, I see nothing which gives a width.  No idea what 'flex:' and 'padding:' are or do either. 

 From my old days writing HTML, seems like there is a line needed to set the 'box' to be 100% (so that it adjusts as the window adjusts - if it is full-screen or in a window you want it to be full-width) - problem is, no idea how to call that.  😥

[image]

Meliox, 2024-02-03:
If you want to play with it, you can get help from chatgpt :)

eremem, 2024-02-25:
@bearhntr Feel free to check the latest version from https://github.com/Meliox/PVE-mods/blob/integration-eremem/pve-mod-gui-temp.sh. It's not quite what you suggested but I hope you find it more readable especially when the summary is shown with 2/3 column-layout.

@Meliox what do you you think of this new layout?

[image]

bearhntr, 2024-02-26:
@eremem 

> @bearhntr Feel free to check the latest version from https://github.com/Meliox/PVE-mods/blob/integration-eremem/pve-mod-gui-temp.sh. It's not quite what you suggested but I hope you find it more readable especially when the summary is shown with 2/3-column layout.

This is amazing work - thanks.   I just installed on one of my PVE systems that has Core i5-3475S (old Dell Optiplex 7010).  I took out the DVD and put in a cover there and it has 2 Kingston SSD (250GB each).  I am thinking that 'sensors' is seeing the 2 ACPI as those (cannot think of anything else it would be seeing)...and the PCI Card is my M.2 WiFi/BT card in a PCIe convertor card.  Guessing that is why the SSD temps are not shown?  

[image]

Meliox, 2024-02-26:
> @bearhntr Feel free to check the latest version from https://github.com/Meliox/PVE-mods/blob/integration-eremem/pve-mod-gui-temp.sh. It's not quite what you suggested but I hope you find it more readable especially when the summary is shown with 2/3-column layout.
> 
> @Meliox what do you you think of this new layout?
> 
> ![obraz](https://private-user-images.githubusercontent.com/56274896/307588772-f0e87727-9e43-454f-bc9f-724e86a799e5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg5ODMwMjQsIm5iZiI6MTcwODk4MjcyNCwicGF0aCI6Ii81NjI3NDg5Ni8zMDc1ODg3NzItZjBlODc3MjctOWU0My00NTRmLWJjOWYtNzI0ZTg2YTc5OWU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI2VDIxMjUyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUzYWUyZGMyODNmMDkwNGJhMTVjMDlmY2NjNmJiY2U2MGE5ZTE2N2NkOWFhZDNhNzgzMDRjYWEwYTE3NDAyMmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8fGUBDvejkeG7Hjmz0x1cIxnSoAwXTRcct8IW02RtiY)

Looks great. I would put nvme and hdd on a new row, aligned with the cpu temp for consistency. So also have more than 4 drives :o

bearhntr, 2024-02-26:
You guys are amazing.  I wish I could read the code better and understand it all.  I still think that the 'info' box (my second image) should be 100% width and the graphs in a 2x2 underneath it.

Meliox, 2024-02-26:
> @eremem
> 
> > @bearhntr Feel free to check the latest version from https://github.com/Meliox/PVE-mods/blob/integration-eremem/pve-mod-gui-temp.sh. It's not quite what you suggested but I hope you find it more readable especially when the summary is shown with 2/3-column layout.
> 
> This is amazing work - thanks. I just installed on one of my PVE systems that has Core i5-3475S (old Dell Optiplex 7010). I took out the DVD and put in a cover there and it has 2 Kingston SSD (250GB each). I am thinking that 'sensors' is seeing the 2 ACPI as those (cannot think of anything else it would be seeing)...and the PCI Card is my M.2 WiFi/BT card in a PCIe convertor card. Guessing that is why the SSD temps are not shown?
> 
> ![image](https://private-user-images.githubusercontent.com/100164602/307956424-17bc79de-709a-4074-9022-185e94376660.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg5ODMwNDcsIm5iZiI6MTcwODk4Mjc0NywicGF0aCI6Ii8xMDAxNjQ2MDIvMzA3OTU2NDI0LTE3YmM3OWRlLTcwOWEtNDA3NC05MDIyLTE4NWU5NDM3NjY2MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyNlQyMTI1NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMTk0MmQyYTQ5N2ZmZWViNTYzNjI0NmI5MmJiMjA5ZGI5YmUxYzcwNzZkODNmNjQ3OWI5YzgzYTc1NTg4NTQ0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.vr_NXvCH2SfqMGvMYljFgHsDQT7-n5u-YeDrKh4pFlo)
> 
> *Notice there are no DRIVES listed:
> 
> ![image](https://private-user-images.githubusercontent.com/100164602/307957172-58e3803f-4a6f-469d-94d5-cf0482db1810.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg5ODMwNDcsIm5iZiI6MTcwODk4Mjc0NywicGF0aCI6Ii8xMDAxNjQ2MDIvMzA3OTU3MTcyLTU4ZTM4MDNmLTRhNmYtNDY5ZC05NGQ1LWNmMDQ4MmRiMTgxMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyNlQyMTI1NDdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iMGU2ZTUxOTNhNzFmMTAyOTM2MWYzNzkzOWQ0ZmZlMTVhZGM1OWFjMWY2OTAwMjE4MWQwODAzNjJhYTA0NDQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.T7RsPS4yO3cuB9Z76jn5cQI030PReeC3ALFnPdp44G8)

Have you tried sensors-detect? Just hit 'enter' for all the questions?

bearhntr, 2024-02-26:
> Have you tried sensors-detect? Just hit 'enter' for all the questions?

What is that? Something I need to install?

bearhntr, 2024-02-26:
I will use a BOOTABLE USB tool - which will give me a Windows way to see if I can figure out what they are.

bearhntr, 2024-02-26:
Well the tool was of no real help - but when I looked at the temps for the 2x SSDs - they were 21°C and 23°C respectively.  Thinking these must have been the drives...but still gonna do more investigating.

eremem, 2024-02-27:
> You guys are amazing. I wish I could read the code better and understand it all. I still think that the 'info' box (my second image) should be 100% width and the graphs in a 2x2 underneath it.

This would be quite a challenge, since we had to move this single box into a new component and inject it before the existing one. We also had to check how if it acts well with multiple columns and their forced number set in the user settings.

eremem, 2024-02-27:
> _The "acpitz-acpi-0" sensor could represent a temperature reading from a specific component on the motherboard, such as the chipset, voltage regulators, or another critical component. However, the exact meaning of this sensor can vary depending on the hardware configuration and the interpretation provided by the motherboard's BIOS or firmware._
> 
> So, i guess you're right about that. A scenario that we have not thought about..
> 
> @eremem: Should we make this configurable using a variable? I am hesitant to add it permanently given the ambiguous usage of acpitz-acpi-0.

@Meliox I wouldn't use it for storage/drive temperature display but maybe as a separate new section e.g. Misc. hardware components (or sth. shorter;)). Than we could enable them (or even group some) via a configuration variable.

Meliox, 2024-02-27:
@eremem No you're right. Does not look good. Have both drive types on a new line and align temps to the right?
Then we can make another "section" for others below as you suggested.

eremem, 2024-02-27:
> @eremem No you're right. Does not look good. Have both drive types on a new line and align temps to the right? Then we can make another "section" for others below as you suggested.

@Meliox Yes, this is the current state:
[image]

This layout would indeed allow us to add further groups down below.

closed 2024-05-11 by eremem