Meliox/PVE-mods #20: sensors not showing under reverse proxy

andrei4002, 2024-03-30:
got a weird issue with showing up the temp sensors when under a reverse proxy

so going to local-network-ip:8006 shows the sensors, but when opening the page using a domain, setup under traefik w/ssl, they don't show up at all, like it's the original page loading without any mods

i don't think it's a browser cache issue since i tried in multiple browsers and all have the same behavior

this is under rev proxy:
[image]

this is under local ip access:
[image]

can't say if it's really an issue with the mod (it doesn't sound like it) but wanted to report it, maybe there's an idea to fix it

Meliox, 2024-03-31:
Hello,
Yes, that's weird. I am not seeing the same as you. It's working well under traefik w. ssl.
[image]

andrei4002, 2024-04-02:
2 days later, they magically started showing up as well. no clue what happened though

closed 2024-04-02 by andrei4002