Meliox/PVE-mods #21: Revert temp mod manually

billyjoebob999, 2024-04-03:
How do you go about reverting the changes manually? I found this mod last night but when I used the script it did not make a backup of my Nodes.pm or pvemanagerlib.js files so I'm unable to use the uninstall using the script.

Meliox, 2024-04-03:
Hmm. That I don't understand. Did you get any errors?

Anyways, there's a couple of options. I have not tried any of those, so please consider backups.

1. You can get the original files by reinstalling the pve-manager package
2. Wait for an update pve-manager  package
3. Files are not directly available from git as one needs to be build. You can get them from the current or older deb package. Get hold of deb package corresponding to your version from http://download.proxmox.com/debian/pve/dists/bookworm/pve-no-subscription/binary-amd64/pve-manager_**8.1.10**_amd64.deb. Download, open the deb-file (7zip works on windows). Get the pve-manager_8.1.10_amd64.deb\data.tar\.\usr\share\pve-manager\js\pvemanagerlib.js and copy to /usr/share/pve-manager/js/pvemanagerlib.js and pve-manager_8.1.10_amd64.deb\data.tar\.\usr\share\perl5\PVE\API2\Nodes.pm to /usr/share/perl5/PVE/API2/Nodes.pm

billyjoebob999, 2024-04-03:
To be honest I dont recall at this point what errors I got. The only error I get now is:
`Preparing mod installation...
ERROR: Can't get value of subfeature temp1_input: Can't read`

Doesn't seem like a big deal and I do in fact have the temp readout working.

[image]

What I was trying to do was remove the mod so I could change it to show the package temp instead but when I ran the uninstall, I was met with:

`Uninstalling the Proxmox VE temperatures display mod...
Restoring modified files...
[warning] No Nodes.pm files found.
[warning] No pvemanagerlib.js files found.`

I did check the folder where those files are supposed to get backed up but it was empty.

Since it works, I will probably just wait for proxmox to update.

closed 2024-04-14 by eremem