Meliox/PVE-mods #22: Proxmox Temp Display Mod - ERROR: Can't get value of subfeature temp1_input: Can't read

borez, 2024-04-06:
I get a script error when running the Proxmox temperature mod. It used to be working until I manually installed IT87 driver to display fan speeds.

This is the error output.

```
Installing the Proxmox VE temperatures display mod...

Preparing mod installation...
ERROR: Can't get value of subfeature temp1_input: Can't read

Detecting support for HDD/SDD temperature sensors...

Detecting support for NVMe temperature sensors...
Detected sensors:
nvme-pci-0300

Detecting support for CPU temperature sensors...
Detected sensor:
coretemp-isa-0000

Backup of "/usr/share/perl5/PVE/API2/Nodes.pm" saved to "/root/backup/Nodes.pm.2024-04-06_22-15-14".
Added thermalstate to /usr/share/perl5/PVE/API2/Nodes.pm.
Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/root/backup/pvemanagerlib.js.2024-04-06_22-15-14".
Expanded space in "/usr/share/pve-manager/js/pvemanagerlib.js".
New temperature display items added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".

Restarting PVE proxy...
```

I have also attached my sensor output:

```
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A  

it8613-isa-0a30
Adapter: ISA adapter
VCore:       726.00 mV (min =  +0.00 V, max =  +2.81 V)
V_SM:          1.20 V  (min =  +0.00 V, max =  +2.81 V)
3VSB:          3.92 V  (min =  +0.00 V, max =  +5.61 V)
+3.3V:         3.34 V  
CPU_fan:     2576 RPM  (min =    0 RPM)
SYS_fan:        0 RPM  (min =    0 RPM)
CPU_temp:     +50.0°C  (low  = -128.0°C, high = +127.0°C)  sensor = thermistor
SYS_temp:     +48.0°C  (low  = -128.0°C, high = +127.0°C)  sensor = thermistor

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +110.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +56.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +54.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +54.0°C  (high = +105.0°C, crit = +105.0°C)
Core 2:        +54.0°C  (high = +105.0°C, crit = +105.0°C)
Core 3:        +54.0°C  (high = +105.0°C, crit = +105.0°C)

nvme-pci-0300
Adapter: PCI adapter
Composite:    +48.9°C  (low  = -273.1°C, high = +69.8°C)
                       (crit = +72.8°C)
Sensor 1:     +48.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +53.9°C  (low  = -273.1°C, high = +65261.8°C)

```

Meliox, 2024-04-06:
ERROR: Can't get value of subfeature temp1_input: Can't read
This error could indicate misconfiguration in sensors. What's is the output of sensors -j? The plugin is using that one. 

Based on your input it should be getting the temperature from despite your relabling.
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +56.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +54.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +54.0°C  (high = +105.0°C, crit = +105.0°C)
Core 2:        +54.0°C  (high = +105.0°C, crit = +105.0°C)
Core 3:        +54.0°C  (high = +105.0°C, crit = +105.0°C)

borez, 2024-04-09:
Thanks for this! For some reason, a reboot of the host worked.
This was referring to the WiFi/Bluetooth card. 

`ERROR: Can't get value of subfeature temp1_input: Can't read`

closed 2024-04-09 by borez