Meliox/PVE-mods #33: Blank screen after refresh directly proceeding install. - PVE 8.2.2

stevensit-admin, 2024-05-10:
Under a relativley fresh install of PVE 8.2.2 the installation causes a blank / black screen after a browser refresh.

note: following a tutorial to achieve a similar result, also resulted in the same issue, could be related, might not be.
         running dual cpu's on Poweredge T420.
         have a NIC with sensor information. (TX401)
        do not know very much about perl or javascript yet, happy to help where I can though.

uninstall worked perfectly and GUI interface returned to working as expected.

What the script accomplishes in the screen shots looks perfect. Feel free to reach out if you want for more info.

eremem, 2024-05-11:
Could you please attach the output of `sensor - j` from your system?

stevensit-admin, 2024-05-11:
Please see attached:
-----------------------------------------------------------
root@nodex:~# sensors -j
{
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 39.000,
         "temp1_max": 87.000,
         "temp1_crit": 97.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 31.000,
         "temp2_max": 87.000,
         "temp2_crit": 97.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 34.000,
         "temp3_max": 87.000,
         "temp3_crit": 97.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 32.000,
         "temp4_max": 87.000,
         "temp4_crit": 97.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 36.000,
         "temp5_max": 87.000,
         "temp5_crit": 97.000,
         "temp5_crit_alarm": 0.000
      },
      "Core 4":{
         "temp6_input": 31.000,
         "temp6_max": 87.000,
         "temp6_crit": 97.000,
         "temp6_crit_alarm": 0.000
      },
      "Core 5":{
         "temp7_input": 39.000,
         "temp7_max": 87.000,
         "temp7_crit": 97.000,
         "temp7_crit_alarm": 0.000
      }
   },
   "enp9s0-pci-0900":{
      "Adapter": "PCI adapter",
      "PHY Temperature":{
         "temp1_input": 84.617
      },
      "MAC Temperature":{
         "temp2_input": 81.825
      }
   },
   "coretemp-isa-0001":{
      "Adapter": "ISA adapter",
      "Package id 1":{
         "temp1_input": 35.000,
         "temp1_max": 87.000,
         "temp1_crit": 97.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 31.000,
         "temp2_max": 87.000,
         "temp2_crit": 97.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 34.000,
         "temp3_max": 87.000,
         "temp3_crit": 97.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 32.000,
         "temp4_max": 87.000,
         "temp4_crit": 97.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 29.000,
         "temp5_max": 87.000,
         "temp5_crit": 97.000,
         "temp5_crit_alarm": 0.000
      },
      "Core 4":{
         "temp6_input": 31.000,
         "temp6_max": 87.000,
         "temp6_crit": 97.000,
         "temp6_crit_alarm": 0.000
      },
      "Core 5":{
         "temp7_input": 29.000,
         "temp7_max": 87.000,
         "temp7_crit": 97.000,
         "temp7_crit_alarm": 0.000
      }
   },
   "nvme-pci-0a00":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 43.850,
         "temp1_max": 76.850,
         "temp1_min": -0.150,
         "temp1_crit": 78.850,
         "temp1_alarm": 0.000
      },
      "Sensor 2":{
ERROR: Can't get value of subfeature temp3_max: I/O error
ERROR: Can't get value of subfeature temp3_min: I/O error
         "temp3_input": 61.850
      }
   }
}

eremem, 2024-05-11:
Please download and test the latest version. You should now be able to see the temperatures for all your CPUs.

The actual problem in this case was an invalid JSON output provided by the `sensors` command, as it bluntly inserts plain-text error messages instead of format them accordingly...  I added some code to sanitize the output so that the other correct readings can now be used anyway.

As far as the NIC temperatures are concerned, at the moment there is no support for sensor categories other than the ones mentioned in the README section.

stevensit-admin, 2024-05-11:
Very nice, works as expected now. Really great turn around. Very appreciative.

Can not do it today, but plan on donating in the future. :)
Again Thank you, great work.

closed 2024-05-11 by stevensit-admin