Meliox/PVE-mods #39: Temperature output missing under 8.2.2

Weishaupt, 2024-05-16:
I'm new to proxmox and installed 8.2.2 freshly from source.

I don't see any output related to temperature in the summary page.

[image]

Installation went trough without isse.

[image]

`sensors -j reads`

```
{
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 41.000,
         "temp1_max": 105.000,
         "temp1_crit": 105.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 39.000,
         "temp2_max": 105.000,
         "temp2_crit": 105.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 39.000,
         "temp3_max": 105.000,
         "temp3_crit": 105.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 39.000,
         "temp4_max": 105.000,
         "temp4_crit": 105.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 39.000,
         "temp5_max": 105.000,
         "temp5_crit": 105.000,
         "temp5_crit_alarm": 0.000
      }
   },
   "acpitz-acpi-0":{
      "Adapter": "ACPI interface",
      "temp1":{
         "temp1_input": 27.800
      }
   },
   "nvme-pci-0500":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 39.850,
         "temp1_max": 82.850,
         "temp1_min": -273.150,
         "temp1_crit": 84.850,
         "temp1_alarm": 0.000
      }
   }
}
```

eremem, 2024-05-16:
The installation returned no errors. If you see no changes, it's most likely because of the browser's cache. Please clear it and reload the web UI.

Meliox, 2024-05-16:
Hi,
Have you tried to fresh you browser properly/clear cache? Ctrl + F5

Weishaupt, 2024-05-16:
Thank you for your reply. Indeed it was the cache. Maybe you could add a message to the script that highlights remembering to clear the browser cache.

closed 2024-05-16 by Weishaupt

Meliox, 2024-05-16:
Thanks for the feedback.. I will add a message to the script this weekend.