Meliox/PVE-mods #41: Missing fan speed

Turtletrumpet, 2024-05-24:
Hello,

first of all, thanks for sharing your great work. I use proxmox at home.
I miss the fan speed:
[image]

```
root@pve ~ > sensors
asus_wmi_sensors-virtual-0
Adapter: Virtual device
CPU Core Voltage:          1.27 V
+12V Voltage:             12.23 V
+5V Voltage:               4.99 V
3VSB Voltage:              3.31 V
CPU Fan:                 1467 RPM
Chassis Fan 1:           1548 RPM
Chassis Fan 2:           1400 RPM
Chassis Fan 3:           1386 RPM
AIO Pump:                   0 RPM
Water Pump:                 0 RPM
CPU OPT:                    0 RPM
CPU Temperature:          +46.0°C
Motherboard Temperature:  +31.0°C
Chipset Temperature:      +45.0°C
Tsensor 1 Temperature:   +216.0°C

nvme-pci-0100
Adapter: PCI adapter
Composite:    +38.9°C  (low  =  -5.2°C, high = +83.8°C)
                       (crit = +87.8°C)

asusec-isa-0000
Adapter: ISA adapter
CPU Core:      1.26 V
CPU_Opt:        0 RPM
Chipset:      +45.0°C
CPU:          +46.0°C
Motherboard:  +31.0°C
T_Sensor:     -40.0°C
VRM:          +20.0°C
CPU:          17.00 A

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +46.2°C
Tccd1:        +39.2°C
```
Do you can fix it? Many thanks!

Dirk

Meliox, 2024-05-24:
@Turtletrumpet: Can you try the changed installer in the PR  #42 and let me know the outcome?

Turtletrumpet, 2024-05-24:
> @Turtletrumpet: Can you try the changed installer in the PR #42 and let me know the outcome?

I hope I use the right file pve-mod-gui-sensors.sh.
The output of the installation:

```
Select a unit for temperature readings...
Type C for Celsius or F for Fahrenheit and press Enter: c

Backup of "/usr/share/perl5/PVE/API2/Nodes.pm" saved to "/root/backup/Nodes.pm.2024-05-24_22-24-11".
Sensors' output added to "/usr/share/perl5/PVE/API2/Nodes.pm".
Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/root/backup/pvemanagerlib.js.2024-05-24_22-24-11".
Expanded space in "/usr/share/pve-manager/js/pvemanagerlib.js".
./pve-mod-gui-sensors.sh: line 557: fan_input: command not found
./pve-mod-gui-sensors.sh: command substitution: line 557: unexpected EOF while looking for matching `}'
./pve-mod-gui-sensors.sh: line 557: ]}: command not found
./pve-mod-gui-sensors.sh: line 557: Fan: command not found
./pve-mod-gui-sensors.sh: line 557: nbsp: command not found
./pve-mod-gui-sensors.sh: line 557: nbsp: command not found
./pve-mod-gui-sensors.sh: line 557: RPM: command not found
Sensor display items added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".

Restarting PVE proxy...
Installation completed

```
After runnig the installation, the web-interface of the node ist hanging.
I attach my used pve-mod-gui-sensors.sh.
Thanks.
[pve-mod-gui-sensors.txt](https://github.com/Meliox/PVE-mods/files/15439710/pve-mod-gui-sensors.txt)

Turtletrumpet, 2024-05-24:
I think, I made a mistake with the transfer of the pve-mod-gui-sensors.sh. I copy the text out of the web-interface and paste it into the editor of winscp. I have not found a way to download the modified file.

Meliox, 2024-05-24:
No, you did not. I am missing to escape something apparently.

Turtletrumpet, 2024-05-25:
Hello,
the scripts runs with no error:
```
root@pve ~ > ./pve-mod-gui-sensors.sh install

Installing the Proxmox VE sensors display mod...

Preparing mod installation...

Detecting support for CPU temperature sensors...
Detected sensors:
k10temp-pci-00c3

Detecting support for HDD/SDD temperature sensors...
Detected sensors:
drivetemp-scsi-1-0
drivetemp-scsi-0-0

Detecting support for NVMe temperature sensors...
Detected sensors:
nvme-pci-0100

Detecting support for fan speed readings...
Detected fan speed sensors:
fan1_input
fan2_input
fan3_input
fan4_input
fan5_input
fan6_input
fan7_input
fan1_input

Select a unit for temperature readings...
Type C for Celsius or F for Fahrenheit and press Enter: C

Backup of "/usr/share/perl5/PVE/API2/Nodes.pm" saved to "/root/backup/Nodes.pm.2024-05-25_14-03-13".
Sensors' output added to "/usr/share/perl5/PVE/API2/Nodes.pm".
Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/root/backup/pvemanagerlib.js.2024-05-25_14-03-13".
Expanded space in "/usr/share/pve-manager/js/pvemanagerlib.js".
Sensor display items added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".

Restarting PVE proxy...
Installation completed
Clear the browser cache to ensure all changes are visualized.
```
but there is no fan displayed after clearing the browser cache:
[image]

I download the script here

[image]

is that the right location?

Meliox, 2024-05-25:
This one https://raw.githubusercontent.com/Meliox/PVE-mods/ddf7274cdab937e61796c46598cbee54003afaac/pve-mod-gui-sensors.sh

Turtletrumpet, 2024-05-25:
thanks for the link.
But, same issue:
[image]
after uninstall -> install -> clear browser cache

Turtletrumpet, 2024-05-25:
I attach the sensor -J output.
[AsusX470.txt](https://github.com/Meliox/PVE-mods/files/15443539/AsusX470.txt)

Meliox, 2024-05-27:
Took a while to figure out how to use your sensor -j output for debuging... Anyhow, figured it out. So here's a new version that recursively finds fanspeeds.

https://raw.githubusercontent.com/Meliox/PVE-mods/75a847979d4e115e8ebd645a05f28a2c5edcfd8c/pve-mod-gui-sensors.sh

Expected output (for fans only)
[image]

Turtletrumpet, 2024-05-28:
> Took a while to figure out how to use your sensor -j output for debuging... Anyhow, figured it out. So here's a new version that recursively finds fanspeeds. new version: https://raw.githubusercontent.com/Meliox/PVE-mods/75a847979d4e115e8ebd645a05f28a2c5edcfd8c/pve-mod-gui-sensors.sh
> 
> Expected output (for fans only) ![image](https://private-user-images.githubusercontent.com/5264368/334199793-c169aa86-7e57-432b-aa19-80e149428df6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTY4ODE4MTcsIm5iZiI6MTcxNjg4MTUxNywicGF0aCI6Ii81MjY0MzY4LzMzNDE5OTc5My1jMTY5YWE4Ni03ZTU3LTQzMmItYWExOS04MGUxNDk0MjhkZjYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUyOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MjhUMDczMTU3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Yzg5OThjMzI0Yjk5ZDVjMzQwMTYyOWVmNDkyZWU4ZTE0YmRkYWQxYzVjZTAwN2JiMGJiOTQ1MGM3MWJiODc0OSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.gq3JuopY_B7zyci5sYDs7AKexVFwYBWtTJO5tpFLhmU)
> 
> So all fans are displayed. Filtering 0 is not really an option as the system can simply have stopped the fan.

Hello, that's some progress. The fan speed appears. However, fan 1 is displayed twice.

I wish that fans with a speed of 0 would not be displayed. I assume that fans in a server are always spinning, right?
[image]

Turtletrumpet, 2024-05-28:
The fan 1 with 1427 rpm is the CPU fan. fan 2 is chassis fan 1.

Meliox, 2024-05-28:
Your system supports several fans. Seamingly 2 different for your cpu.

About removing fans with speed=0. Well quite many run smaller servers where the cpu only spins up of the temperature is high enough. That means removing them from displays is inaccurate information.

So instead of fans you would like to see cpu fan, chassis etc?

Turtletrumpet, 2024-05-29:
That sounds promising. I'll take a look and report back. Thanks!

Meliox, 2024-06-11:
@Turtletrumpet: I will merge the code as is now as if fixes the original problem. If your investigation turns out fruitfull, let me know. Thanks

Meliox, 2024-06-16:
@Turtletrumpet: I've added an option to display or not display non active fans. If a idle fan, e.g. speed=0, spins up, then it will be displayed.

closed 2024-06-16 by Meliox