Meliox/PVE-mods #45: Add extra host info (hardware models / serials)

ITJamie, 2024-05-30:
It would be great if extra hardware model info could be displayed.

manufacturer
model
serial

these can be parsed out of demidecode pretty easily

```
dmidecode -t 1
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 2.8 present.

Handle 0x0100, DMI type 1, 27 bytes
System Information
        Manufacturer: Dell Inc.
        Product Name: PowerEdge R630
        Version: Not Specified
        Serial Number: FAKESERIAL
        UUID: FAKEUUID
        Wake-up Type: Power Switch
        SKU Number: SKU=NotProvided;ModelName=PowerEdge R630
        Family: Not Specified
```

Meliox, 2024-05-30:
May I ask why manufacturer and product is of interest for you?

ITJamie, 2024-05-30:
We have a lot of different hardware providers. Some of the model numbers on asrock or supermicro boxes dont make it in anyway obvious at first glance what the server type is

Meliox, 2024-05-30:
Ok, could you list what information would be needed?

Suggestion for where it could be? https://raw.githubusercontent.com/Meliox/PVE-mods/main/pve-mod-sensors.png

Also, is dmidecode a standard Linux tool available in proxmox? (don't have access right now)

eremem, 2024-05-30:
@Meliox `dmidecode` seems to be available by default on a PVE installation.

eremem, 2024-06-05:
@Meliox I guess the data provided by dmidecode in this use-case could be considered static, so the installer script, which is run with root privileges, could dump the output of dmidecode to a file in e.g. /usr/share/pve-manager/js/ and ideally also transform it to some more "digestible" format to avoid later conversions in runtime. The perl script could than just `cat` the dump file and make it available to the JS code via some new variable. How do you find such an approach?

Meliox, 2024-06-05:
@eremem: That's a very good work around.  Yes, that can easily be done.

Meliox, 2024-06-05:
[image]

(nb. information not available on my system)

Maybe move to above Kernal Version?

eremem, 2024-06-05:
@Meliox I'm getting the following error during the installation:
```
./pve-mod-gui-sensors.sh: line 260: syntax error near unexpected token `('
./pve-mod-gui-sensors.sh: line 260: `        title: gettext('System Information'),'
```

Meliox, 2024-06-05:
@eremem yes... The PR is currently broken. I left the js code hanging until decision on where it should be placed.

eremem, 2024-06-05:
> ![image](https://private-user-images.githubusercontent.com/5264368/336965746-efb4be96-560f-449b-a52f-ffbcbd70443d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTc2MTc4MzEsIm5iZiI6MTcxNzYxNzUzMSwicGF0aCI6Ii81MjY0MzY4LzMzNjk2NTc0Ni1lZmI0YmU5Ni01NjBmLTQ0OWItYTUyZi1mZmJjYmQ3MDQ0M2QucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDYwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA2MDVUMTk1ODUxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Y2EyMDdlMTc5M2U4MGZiMzk4OWVkM2Q3OTFmMWI1MzkwMTQwYTg5YjVjNWM1MGRiN2UyNDQ5NzUyNGJmODYyMSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.Dd2PRs17arut9ZEGN3r_B6HYbsyRy6c-VXIBXGYtPVI)
> 
> (nb. information not available on my system)
> 
> Maybe move to above Kernal Version?
> 
> Let me know what you think @eremem, @ITJamie .

My first thought was about moving it before the kernel version so 👍 to that! 😄

eremem, 2024-06-05:
Works and looks good:
[image]

closed 2024-06-15 by Meliox