Meliox/PVE-mods #48: No temp on AMD Embedded G-Series GX-420GI Radeon R7E

Undercore90, 2024-06-07:
Hi,

Can you please help me to fix this error. It's probably a matter of correcting some file. Until the patch is released, I can edit it manually.

fam15h_power-pci-00c4
Adapter: PCI adapter
power1:       42.04 mW (avg =  67.69 mW, interval =   0.01 s)
                       (crit =  15.00 W)

amdgpu-pci-0008
Adapter: PCI adapter
vddgfx:      887.00 mV 
vddnb:       887.00 mV 
edge:         +48.0°C  

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +48.8°C  (high = +70.0°C)
                       (crit = +100.0°C, hyst = +99.0°C)

[image]

Meliox, 2024-06-07:
Please provide the sensor -j output. Thanks.

Undercore90, 2024-06-07:
Sure,

```
root@pve:~# sensors -j
{
   "fam15h_power-pci-00c4":{
      "Adapter": "PCI adapter",
      "power1":{
         "power1_average": 0.028,
         "power1_input": 0.033,
         "power1_crit": 15.000,
         "power1_average_interval": 0.010
      }
   },
   "amdgpu-pci-0008":{
      "Adapter": "PCI adapter",
      "vddgfx":{
         "in0_input": 0.887
      },
      "vddnb":{
         "in1_input": 0.887
      },
      "edge":{
         "temp1_input": 49.000
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "temp1":{
         "temp1_input": 49.250,
         "temp1_max": 70.000,
         "temp1_crit": 100.000,
         "temp1_crit_hyst": 99.000
      }
   }
}
```

Undercore90, 2024-06-11:
Hi,

Any news here?

Meliox, 2024-06-11:
Your AMD system is reporting temp differently. Can you try the version provided in #49 . Please uninstall first, then install the new one.

Undercore90, 2024-06-11:
How to install #49 version?

Undercore90, 2024-06-12:
[image]
Ok something worked out I need to change the name again e.g. core and it will be fine.

Undercore90, 2024-06-12:
Hi,

I did it this way:
cd /tmp
git clone https://github.com/Meliox/PVE-mods.git
cd PVE-mods
git fetch origin pull/49/head:pr-49
git checkout pr-49
bash pve-mod-gui-sensors.sh install

It didn't recognize my CPU it displayed sensors -j and asked where the CPU temperature was.

Meliox, 2024-06-12:
Thanks for testing. Found the missing code line. Can I ask you to try again?

PR #49 (2024-06-16): Add support for multiple AMD temp types
Fixes #48 
Changes add support for AMDs 2 temperature reporting variants (temp and Tccd). Example:

```
root@pve:~# sensors -j
{
   "fam15h_power-pci-00c4":{
      "Adapter": "PCI adapter",
      "power1":{
         "power1_average": 0.028,
         "power1_input": 0.033,
         "power1_crit": 15.000,
         "power1_average_interval": 0.010
      }
   },
   "amdgpu-pci-0008":{
      "Adapter": "PCI adapter",
      "vddgfx":{
         "in0_input": 0.887
      },
      "vddnb":{
         "in1_input": 0.887
      },
      "edge":{
         "temp1_input" [... cut]
closed 2024-06-15 by Undercore90