Meliox/PVE-mods #63: Change how often information is refreshed

KittyFarts, 2024-06-24:
**Is your feature request related to a problem? Please describe.**
Not really a problem but is causing I/O delay in Proxmox (about 2-5% increase)

**Describe the solution you'd like**
Refresh Manually or at an interval (ie Auto, 5 seconds, 10 seconds, 15 seconds, etc.)

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**Sensors output**
N/A

Thank you!

Meliox, 2024-06-24:
Hi,

Since you're the first one reporting I/O delays. Could you share a screenshot proxmox of before/after using the mod?

KittyFarts, 2024-06-25:
Hi there!

I just did a fresh reinstall of the latest Proxmox yesterday on the same unit. Without the mod I see it in the low 0.10% but when I install the temperature mod I am seeing 1.0%+

I am seeing the I/O Delay at around 1-2% on this fresh install - same number of VMs (3 - Pfsense, UniFi Console, and TrueNAS) with same processes running inside those VMs.

Not that it's a big deal, however, having it refresh every x seconds would be awesome. Sometimes people have higher I/O delays that I do and that 2%+ might mean more.

[image]
[image]
[image]
[image]