Meliox/PVE-mods #64: Get N/A in CPU Thermal State

hugosbnarciso, 2024-06-27:
**Describe the bug**
Get N/A in CPU Thermal State

**To Reproduce**
Fresh Install

**Expected behavior**
CPU Temp

**Screenshots**
[image]

**Sensors output**
`root@pve:~# sensors -j
{
   "iwlwifi_1-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
ERROR: Can't get value of subfeature temp1_input: Can't read

      }
   },
   "nvme-pci-0300":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 44.850,
         "temp1_max": 89.850,
         "temp1_min": -273.150,
         "temp1_crit": 94.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 44.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      },
      "Sensor 2":{
         "temp3_input": 61.850,
         "temp3_max": 65261.850,
         "temp3_min": -273.150
      }
   },
   "amdgpu-pci-0400":{
      "Adapter": "PCI adapter",
      "vddgfx":{
         "in0_input": 1.218
      },
      "vddnb":{
         "in1_input": 0.812
      },
      "edge":{
         "temp1_input": 58.000
      },
      "PPT":{
         "power1_input": 10.000
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 63.750
      }
   }
}`

Meliox, 2024-06-27:
Thanks for the report. It seems like AMD is very inconsistent with how they are reporting the temperatures...

hugosbnarciso, 2024-06-27:
i understand. let me know if you need more info or testing from my side, happy to help

Meliox, 2024-06-27:
Thanks. You provided everything needed. If you need a temporary fix, then you should be able change Tccd to Tctl  https://github.com/Meliox/PVE-mods/blob/44ff3dccb3edf0843af4dc35dcd353137fd53557/pve-mod-gui-sensors.sh#L157 to

hugosbnarciso, 2024-06-27:
that works, thanks !

Meliox, 2024-06-27:
Happy to hear that.. I will look into a permanent solution later.

DarrenOfficial, 2024-06-30:
> You provided everything needed. If you need a temporary fix, then you should be able change Tccd to Tctl

Thanks 2! this works for me as well. Ryzen 5 5500 CPU

[image]

Meliox, 2024-06-30:
@DarrenOfficial can I ask you also to provide your sensor -j output here for comparison. Thank you.

DarrenOfficial, 2024-07-03:
Sure, @Meliox 

```
~# sensors -j
{
   "drivetemp-scsi-1-0":{
      "Adapter": "SCSI adapter",
      "temp1":{
         "temp1_input": 39.000,
         "temp1_max": 40.000,
         "temp1_min": 10.000,
         "temp1_crit": 60.000,
         "temp1_lcrit": 5.000,
         "temp1_lowest": 37.000,
         "temp1_highest": 51.000
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 63.250
      }
   },
   "nvme-pci-0600":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 36.850,
         "temp1_max": 82.850,
         "temp1_min": -5.150,
         "temp1_crit": 84.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 36.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      }
   },
   "drivetemp-scsi-0-0":{
      "Adapter": "SCSI adapter",
      "temp1":{
         "temp1_input": 38.000,
         "temp1_max": 40.000,
         "temp1_min": 10.000,
         "temp1_crit": 60.000,
         "temp1_lcrit": 5.000,
         "temp1_lowest": 38.000,
         "temp1_highest": 56.000
      }
   },
   "nvme-pci-0400":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 39.850,
         "temp1_max": 84.850,
         "temp1_min": -0.150,
         "temp1_crit": 94.850,
         "temp1_alarm": 0.000
      }
   }
}

```

PR #66 (2024-07-05): Add support for one more AMD variant
Fixes #64 

Supports:

1)
```bash
{
   "drivetemp-scsi-1-0":{
      "Adapter": "SCSI adapter",
      "temp1":{
         "temp1_input": 39.000,
         "temp1_max": 40.000,
         "temp1_min": 10.000,
         "temp1_crit": 60.000,
         "temp1_lcrit": 5.000,
         "temp1_lowest": 37.000,
         "temp1_highest": 51.000
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 63.250
      }
   },
   "nvme-pci-0600":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 36.85 [... cut]
commit 678ae108 (2024-07-05): Add support for one more AMD variant (#66)

Fixes #64
closed 2024-07-05 by Meliox