Meliox/PVE-mods #67: Invalid entry length (0). DMI table is broken! Stop.

paf2332, 2024-07-10:
When running the bash pve-mod-gui-sensors.sh install
I get an error 
Invalid entry length (0). DMI table is broken! Stop.
after the Sensors' output added to "/usr/share/perl5/PVE/API2/Nodes.pm".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
**Sensors output**
If applicable, provide sensor datadump (use the argument save-sensors-data)

**Additional context**
Add any other context about the problem here.

Meliox, 2024-07-10:
What is the result when executing?
```dmidecode -t 1```

paf2332, 2024-07-10:
Hi 

 

Thanks for the reply

 

I’m running on an anonymous Chinese mini PC which looks as though it doesn’t return much information!

 

 

 

Paul Farrant

Meliox, 2024-07-10:
Please share the output.

paf2332, 2024-07-10:
dmidecode -t 1
# dmidecode 3.4
Getting SMBIOS data from sysfs.
SMBIOS 3.5.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
        Manufacturer: Default string
        Product Name: Default string
        Version: Default string
        Serial Number: Default string
        UUID: 03000200-0400-0500-0006-000700080009
        Wake-up Type: Power Switch
        SKU Number: Default string
        Family: Default string

Invalid entry length (0). DMI table is broken! Stop.

Meliox, 2024-07-10:
Okay.. Well you have to ask your manufacturers to make a new bios with fixes for that.

You can disable system information with a no when asked during install. It does not bring any value to show anyhow.

Meliox, 2024-07-11:
Closing this as it is an expected output by dmidecode in case of a broke table, hence it makes sense to keep it.
Workaround is to disable system information.

closed 2024-07-11 by Meliox