Meliox/PVE-mods #68: unable to unistall gui-sensors

radu-boboc, 2024-08-24:
First of all,  Thank you very much for these scripts!!

**Describe the bug**
After a lot of work I managed to make my fan show in sensors and I wanted it displayed in gui so I uninstalled the original senors script: 
./pve-mod-gui-sensors.sh uninstall
and then reinstall it with some new options like show averages and display fans and so on.
Not only that it didn't work, but now I can't restore my original files because the for some unkown reason the backup already contains  the modified gui.
So now I am in a state that the cpu temps are displayed ( one for each core), the drive temp as well, but not the fans.
This is not a request on how to fix this, it is simply an issue on restoring the original gui.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to shell
2. ./pve-mod-gui-sensors.sh uninstall
3. Reload page
4. Observe that the restored gui is actually the current one

**Expected behavior**
I expect to have the original gui
[image]
This is the "restored" gui
**Sensors output**
[image]

**Additional context**
updated Virtual Environment 8.2.4

I am guessing that installing all your scripts created multiple backups and the original gui is lost.

Thank you for this scripts!

radu-boboc, 2024-08-24:
just for clarity I'll paste the install options:
[image]

and the output ( that doesn't match what I choose above, for example average on cores):
[image]

eremem, 2024-08-24:
How many backup files for pvemanagerlib.js are there in your /root/backup directory? If there are multiple, going backwards based on the timestamps, does any of them contain the content without the modifications? 
In the screenshot from your installation there are no errors shown. At the end one can see where the original files were copied. The installer script can detect if the modifications are already in place and stops, so none of the backup files should contain any customizations. Did you edit the modified files in /usr/share/... after the installation and tried to install the mods again?

radu-boboc, 2024-08-24:
[image]

I did not edit the files in /usr/share/...
I have several backups, and I will now try to restore the older one.
Thanks for the idea!

closed 2025-01-06 by Meliox