Meliox/PVE-mods #69: Add iGPU usage stats

Unspec7, 2024-09-14:
**Is your feature request related to a problem? Please describe.**
I'm currently using frigate in an unprivileged LXC, and so I cannot see iGPU stats. I was wondering if iGPU stats could be included in the CPU/drive/etc card?

**Describe the solution you'd like**
Able to see GPU utilization percentages

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

**Sensors output**

From "intel_gpu_top -J":

`{
        "period": {
                "duration": 36.071383,
                "unit": "ms"
        },
        "frequency": {
                "requested": 0.000000,
                "actual": 0.000000,
                "unit": "MHz"
        },
        "interrupts": {
                "count": 0.000000,
                "unit": "irq/s"
        },
        "rc6": {
                "value": 99.998680,
                "unit": "%"
        },
        "power": {
                "GPU": 0.000000,
                "Package": 4.944216,
                "unit": "W"
        },
        "imc-bandwidth": {
                "reads": 624.670112,
                "writes": 59.454119,
                "unit": "MiB/s"
        },
        "engines": {
                "Render/3D/0": {
                        "busy": 0.000000,
                        "sema": 0.000000,
                        "wait": 0.000000,
                        "unit": "%"
                },
                "Blitter/0": {
                        "busy": 0.000000,
                        "sema": 0.000000,
                        "wait": 0.000000,
                        "unit": "%"
                },
                "Video/0": {
                        "busy": 0.000000,
                        "sema": 0.000000,
                        "wait": 0.000000,
                        "unit": "%"
                },
                "VideoEnhance/0": {
                        "busy": 0.000000,
                        "sema": 0.000000,
                        "wait": 0.000000,
                        "unit": "%"
                }
        },
        "clients": {
                "4293430945": {
                        "name": "ffmpeg",
                        "pid": "1536351",
                        "engine-classes": {
                                "Render/3D": {
                                        "busy": "0.000000",
                                        "unit": "%"
                                },
                                "Blitter": {
                                        "busy": "0.000000",
                                        "unit": "%"
                                },
                                "Video": {
                                        "busy": "0.099742",
                                        "unit": "%"
                                },
                                "VideoEnhance": {
                                        "busy": "0.000000",
                                        "unit": "%"
                                }
                        }
                }
        }
}`

divStar, 2024-11-20:
Not sure about the iGPU part, but I use `telegraf` for CPU (and other host-related) monitoring and pass it into an InfluxDB, that I then can query at will or have the data be sent or pulled by grafana. Telegraf doesn't support iGPU monitoring though from what I know, but I found this (for Unraid though): https://forums.unraid.net/topic/93302-support-intel-igpu-utilization-stats-into-influxdb-for-use-with-grafana-intel-gpu-telegfraf/ . Perhaps it could make sense to investigate how it's done.

`lm-sensors` does not really work much in my case and I also don't have any GPU (apart of that on my server motherboard) in my system, so no way to test or really assist much.

Meliox, 2025-01-06:
Unfortunlate, I do not think it is possible to retrieve GPU utilisation with a non-elevated user using "intel_gpu_top -J". User interface is running as www-data and as far as I can find is that it only works under root for me.

Meliox, 2025-01-06:
@Unspec7: Hmm upon further investigation, I think I managed to find a workaround. Though it requires a script running in the background pulling and saving data to the memory/disk. The first will reset data upon reboot. An hour or so is just a couple of hundred bytes. It's not pretty, but does the job.

Pulling the information adds some _non_-significant CPU load. An example below is if pulling every second. Maybe not something that is noticable...

```
PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
3063011 root      20   0    5624   3072   2816 S   2.3   0.0   0:01.06 intel_gpu_top
```

First example is current GPU util:
[image]

I see that the graphs in proxmox only has a minut scale, in order to comply with this an average over 60 seconds can be calculated. I think it's better to make a gpu graph only.

Unspec7, 2025-01-06:
I think in most cases, when the iGPU is being used, users won't be using it for brief moments. For my application at least, it'll be Plex transcodes, so it'll be a long running iGPU load. So I think pulling every second is not necessary - but maybe could be configurable when installing?

Meliox, 2025-01-07:
Well, yes, could make it configurable.

That is not correct. For plex the load does vary a lot and is definitely not a sustained one due to buffering. E.g. it might run 100% for 15 seconds (until fully buffered), and 0 for the 45 seconds. That would give a load of 25%. That means if we just update once a minute, then you would see the average load over that. It would be acceptable in the graph.

The picture above is presenting load every second (as CPU is). That may be valuable as well. Could even make the load 1 min, 5 min, 15 min.

Anyhow, configurable would be nice.