Meliox/PVE-mods #70: Will the script survive Proxmox upgrades?

logicito, 2024-09-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

**Sensors output**
If applicable, provide sensor datadump (use the argument save-sensors-data)

Meliox, 2024-09-25:
I think you forgot the question apart from what can be read in the header...
No, it will not survive an update. I believe the package is pve-manager. If that package is updated the script needs to be reinstalled. Please note that proxmox may introduce breaking chances at any time and thus is can break the script.

logicito, 2024-09-25:
Thank you, I appreciate the quick response

divStar, 2024-11-20:
Just checked with Proxmox VE 8.2.9 (it just had a `pve-manager` update, too) and indeed the script had to be re-applied, but it worked without any issues. I even wrote a Terraform configuration project to prepare my host for further use (I haven't shared it yet), but in that project it downloads the anti-nag script and installs it upon applying the configuration and uninstalls it upon destroying the configuration (amongst many other things).

TL;DR: works great, thanks!

P.s.: I am not sure, but perhaps it'd be possible to create some recurring way of automatically applying the script, but since if the script fails the whole UI fails, I wouldn't really suggest it.

closed 2025-01-06 by Meliox