Meliox/PVE-mods #71: PVE 8.2 - temperature script killed pve proxy - making my server  inaccessible via the web interface

freaked1234, 2024-10-13:
**Describe the bug**
pveproxy rendered unable to start - no web access to the server possible, no restart of pveproxy possible due to wrong syntax in nodes file

**To Reproduce**
Steps to reproduce the behavior:
1. Install lm-sensors
2. run latest script from this github
3. error error error

**Expected behavior**
Stuff works
PVE accessible through web interface
no error error error
no syntax error in nodes file

**Screenshots**
[image]

Okay, after a huge shock and a reminder to backup my whole proxmox config I used putty to connect to my PVE instance and tried to restart pveproxy manually. This gave me an error message, telling me that there is a wrong syntax in line 489 of node.js. Used nano to check the line and found the syntax you see in the screenshot. Instead of 'sensors' I got \sensors\  
After fixing this and manually restarting pveproxy everything is working.

I dont know enough about this stuff to be able to tell what went wrong but I am using german keyboard layout and on that the button for " ' " produces " \ " .

Meliox, 2024-10-13:
Thanks for the report.

Good that you were able to resolve it yourself. Before the script makes modifications, it makes a backup of the files to be edited. In the same directory of where you downloaded the script you will be able to find the original files in the backup folder. You could also simple revert everything by passing uninstall as argument to the script.

You may very well be the first using the script with non-English language. It could very well be that which is causing issues. 

If you are interested in helping then you can perform an uninstall and then add this "export LANG=en_US.UTF-8" in the second line in the script, save it, and report back on the result from Nodes.pm.

The English locale must be installed/available prior to use, you can check that with "locale - a" in your terminal.

freaked1234, 2024-10-13:
I will gladly try tomorrow but it looks like the backup also did not work or I screwed it up by trying to install again before checking via nano.
Is the rest of the following lines also wrong and could you maybe post these lines +10 more above and below from a working config so I can compare the syntax?

Meliox, 2024-10-13:
If you manage to install it on top of existing installation then a new backup file will be created. So look for the oldest one in the backup folder.

Unfortunately, I do not have access to a computer right now.

freaked1234, 2024-10-13:
Ok I will check but it might be 1-2 days until I find the time. Maybe someone can post this part of the config in the meantime just to be safe.

Dexy2811, 2024-11-18:
i had the same issue with the script but ended up fixing it by reinstalling proxmox-widget-toolkit also my locale -a is:
```
root@homelab:~# locale -a
C
C.utf8
en_US.utf8
POSIX
```

feel free to ask if there are any files you wish me to upload.

Unspec7, 2024-12-02:
On the latest version of proxmox with everything updated, I did not have the issue. Proxy started fine.

closed 2025-01-18 by Meliox