Meliox/PVE-mods #72: nothing displayed on promox 8.3.0

mohataj, 2024-12-04:
**Describe the bug**
after installation nothing is displayed

**To Reproduce**
root@proxmox:/usr/share/perl5/PVE/API2# bash pve-mod-gui-sensors.sh install

Installing the Proxmox VE sensors display mod...

Preparing mod installation...

Detecting support for CPU temperature sensors...
Detected sensors:
coretemp-isa-0000

Do you wish to display temperatures for all cores [C] or just an average temperature per CPU [a]? (C/a):
[info] Temperatures will be displayed for all cores.

Detecting support for HDD/SDD temperature sensors...
Detected sensors:
drivetemp-scsi-5-0
drivetemp-scsi-2-0
drivetemp-scsi-3-0

Detecting support for NVMe temperature sensors...
[warning] No NVMe temperature sensors found.

Detecting support for fan speed readings...
[warning] No fan speed sensors found.

Do you wish to display temperatures in degrees Celsius [C] or Fahrenheit [f]? (C/f):
[info] Temperatures will be presented in degrees Celsius.

Do you wish to enable system information? (Y/n): Y
[info] System information will be displayed...

Backup of "/usr/share/perl5/PVE/API2/Nodes.pm" saved to "/usr/share/perl5/PVE/API2/backup/Nodes.pm.2024-12-04_11-56-39".
Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/usr/share/perl5/PVE/API2/backup/pvemanagerlib.js.2024-12-04_11-56-39".
Sensors' output added to "/usr/share/perl5/PVE/API2/Nodes.pm".
System information output added to "/usr/share/perl5/PVE/API2/Nodes.pm".
Expanded space in "/usr/share/pve-manager/js/pvemanagerlib.js".
Sensor display items added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".

Restarting PVE proxy...
Installation completed.
[info] Clear the browser cache to ensure all changes are visualized.

root@proxmox:/usr/share/perl5/PVE/API2# reboot

**Expected behavior**
Temperature should be displayed

**Screenshots**
[image]

**Sensors output**
root@proxmox:~# sensors
gigabyte_wmi-virtual-0
Adapter: Virtual device
temp1:        +44.0°C
temp2:        +49.0°C
temp3:        +48.0°C
temp5:        +61.0°C
temp6:        +46.0°C

drivetemp-scsi-5-0
Adapter: SCSI adapter
temp1:        +41.0°C  (low  =  +0.0°C, high = +60.0°C)
                       (crit low = -41.0°C, crit = +85.0°C)
                       (lowest = +21.0°C, highest = +50.0°C)

drivetemp-scsi-2-0
Adapter: SCSI adapter
temp1:        +46.0°C  (low  =  +0.0°C, high = +70.0°C)
                       (crit low =  +0.0°C, crit = +70.0°C)
                       (lowest = +30.0°C, highest = +56.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C
temp2:        +16.8°C
temp3:        +27.8°C

nouveau-pci-0400
Adapter: PCI adapter
GPU core:    912.00 mV (min =  +0.80 V, max =  +1.19 V)
temp1:        +50.0°C  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)

pch_cometlake-virtual-0
Adapter: Virtual device
temp1:        +55.0°C

drivetemp-scsi-3-0
Adapter: SCSI adapter
temp1:        +46.0°C  (low  =  +0.0°C, high = +65.0°C)
                       (crit low = -40.0°C, crit = +70.0°C)
                       (lowest = +43.0°C, highest = +46.0°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +49.0°C  (high = +84.0°C, crit = +100.0°C)
Core 0:        +48.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:        +45.0°C  (high = +84.0°C, crit = +100.0°C)
Core 2:        +47.0°C  (high = +84.0°C, crit = +100.0°C)
Core 3:        +47.0°C  (high = +84.0°C, crit = +100.0°C)
Core 4:        +49.0°C  (high = +84.0°C, crit = +100.0°C)
Core 5:        +47.0°C  (high = +84.0°C, crit = +100.0°C)

**Additional context**
Add any other context about the problem here.

Meliox, 2024-12-04:
Did you clear your browser cache.? Usually ctrl+f5.

mohataj, 2024-12-04:
it is working
in a private window it is now displayed, cache reset on safari did not work so I have to wait some days

closed 2024-12-04 by mohataj