Meliox/PVE-mods #73: Add RAM Temperature

gluca, 2025-01-02:
lm-sensors provides also memopry (RAM) temperature. Can you please add it in a future release?
Here my json output from sensors -j:
here my sensors -j output. Ram temperature is in the SODIMM section
root@t7910:~# sensors -j
{
   "coretemp-isa-0001":{
      "Adapter": "ISA adapter",
      "Package id 1":{
         "temp1_input": 45.000,
         "temp1_max": 75.000,
         "temp1_crit": 85.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 37.000,
         "temp2_max": 75.000,
         "temp2_crit": 85.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 38.000,
         "temp3_max": 75.000,
         "temp3_crit": 85.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 40.000,
         "temp4_max": 75.000,
         "temp4_crit": 85.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 39.000,
         "temp5_max": 75.000,
         "temp5_crit": 85.000,
         "temp5_crit_alarm": 0.000
      },
      "Core 4":{
         "temp6_input": 39.000,
         "temp6_max": 75.000,
         "temp6_crit": 85.000,
         "temp6_crit_alarm": 0.000
      },
      "Core 5":{
         "temp7_input": 38.000,
         "temp7_max": 75.000,
         "temp7_crit": 85.000,
         "temp7_crit_alarm": 0.000
      }
   },
   "dell_smm-isa-0000":{
      "Adapter": "ISA adapter",
      "Processor Fan":{
         "fan1_input": 995.000,
         "fan1_min": 0.000,
         "fan1_max": 5755.000
      },
      "Processor Fan":{
         "fan2_input": 994.000,
         "fan2_min": 0.000,
         "fan2_max": 5755.000
      },
      "Other Fan":{
         "fan3_input": 600.000,
         "fan3_min": 0.000,
         "fan3_max": 5000.000
      },
      "CPU":{
         "temp1_input": 40.000
      },
      "CPU":{
         "temp2_input": 45.000
      },
      **"SODIMM":{
         "temp3_input": 25.000
      },
      "SODIMM":{
         "temp4_input": 33.000
      },
      "SODIMM":{
         "temp5_input": 33.000
      },
      "SODIMM":{
         "temp6_input": 24.000
      }**
   },
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 40.000,
         "temp1_max": 75.000,
         "temp1_crit": 85.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 36.000,
         "temp2_max": 75.000,
         "temp2_crit": 85.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 35.000,
         "temp3_max": 75.000,
         "temp3_crit": 85.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 35.000,
         "temp4_max": 75.000,
         "temp4_crit": 85.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 36.000,
         "temp5_max": 75.000,
         "temp5_crit": 85.000,
         "temp5_crit_alarm": 0.000
      },
      "Core 4":{
         "temp6_input": 36.000,
         "temp6_max": 75.000,
         "temp6_crit": 85.000,
         "temp6_crit_alarm": 0.000
      },
      "Core 5":{
         "temp7_input": 35.000,
         "temp7_max": 75.000,
         "temp7_crit": 85.000,
         "temp7_crit_alarm": 0.000
      }
   }
}

Meliox, 2025-01-02:
Thanks. Would average or individual temps be most appealing?

gluca, 2025-01-02:
As per CPU, if possible to be chosen during installation
*Gianluca Magalotti*

*Togaf 9 FoundationCobit 5 FoundationITIL V3 2011 Foundation*

Email:    ***@***.***
Mobile:   +39 3474 3474 25
Linkedin: http://www.linkedin.com/in/gianlucamagalotti

closed 2025-01-18 by Meliox