Meliox/PVE-mods #74: Does not work on Proxmox 8.3.2

designworld-ca, 2025-01-04:
**Describe the bug**
- install using wget
- restart server

**To Reproduce**
Steps to reproduce the behavior:
1. open a browser that has not connected to the prox web site before
2. go to proxmox web site
3. Nag screen shows up

**Expected behavior**
Expected no nag screen

**Screenshots**

[image]

Meliox, 2025-01-04:
Hey.
I am currently using 8.3.0, so can I ask you to share the following files:
"/usr/share/javascript/proxmox-widget-toolkit/proxmoxlib.js"
"/usr/share/javascript/proxmox-widget-toolkit/proxmoxlib.min.js"

Before applying modifications. You may need to run uninstall first.

designworld-ca, 2025-01-04:
Hi There
Thanks for your quick reply.  I attach the files.  Just to make sure I did not miss something what is the syntax to uninstall?
[proxmoxlib.min.js.txt](https://github.com/user-attachments/files/18308263/proxmoxlib.min.js.txt)
[proxmoxlib.js.txt](https://github.com/user-attachments/files/18308264/proxmoxlib.js.txt)

Meliox, 2025-01-07:
@Arigoldzld any errors while installing?

Arigoldzld, 2025-01-07:
> @Arigoldzld any errors while installing? 

No, i followed the install via console, nothing weird.

Meliox, 2025-01-07:
@Arigoldzld: Can you look into: #76.

Meliox, 2025-01-07:
@designworld-ca 
I just check. Script still works as intended on 8.3.2. Your attached files indicate that it is not removed (or uninstalled), which is why you see it. 

Could you run the script again with install. Log out of proxmox, refresh your browser with CTRL+F5 (important!) and then log in again. If you still see the message, please share theproxmoxlib.js file again. Thanks.

designworld-ca, 2025-01-07:
I see the issue here.  The install instructions are not complete.  
I had to chmod +x the pve-mod-nag_screen.sh and I had not run the script with the install flag
For sake of completeness you might want to add this line to your install instructions.  Thank you!
./pve-mod-nag-screen.sh install

closed 2025-01-07 by designworld-ca

Meliox, 2025-01-07:
Thanks for confirming. What user is you executing the script with?

Yes, I can update the instructions.

designworld-ca, 2025-01-08:
I was running as root even I though I know it's bad practice, thanks!