Meliox/PVE-mods #80: CPU temperature is displayed as N/A (Ryzen 9 5950X)

MusclePr, 2025-02-01:
**Is your feature request related to a problem? Please describe.**
Please, add support for AMD Ryzen 9 5950X.

**Describe the solution you'd like**
`N/A` ---> `43.75°C`

**Describe alternatives you've considered**
`N/A` ---> `Tctl: 43.75°C (Tccd1: 44.5°C | Tccd2: 39.0°C)`

**Additional context**
Screen shot:
[image]

**Sensors output**
```json
{
   "enp4s0-pci-0400":{
      "Adapter": "PCI adapter",
      "PHY Temperature":{
         "temp1_input": 73.000
      },
      "MAC Temperature":{
         "temp2_input": 73.000
      }
   },
   "nvme-pci-0100":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 51.850,
         "temp1_max": 81.850,
         "temp1_min": -273.150,
         "temp1_crit": 84.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 51.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      },
      "Sensor 2":{
         "temp3_input": 59.850,
         "temp3_max": 65261.850,
         "temp3_min": -273.150
      }
   },
   "nouveau-pci-0600":{
      "Adapter": "PCI adapter",
      "temp1":{
         "temp1_input": 32.000,
         "temp1_max": 95.000,
         "temp1_max_hyst": 3.000,
         "temp1_crit": 105.000,
         "temp1_crit_hyst": 5.000,
         "temp1_emergency": 135.000,
         "temp1_emergency_hyst": 5.000
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 43.750
      },
      "Tccd1":{
         "temp3_input": 44.500
      },
      "Tccd2":{
         "temp4_input": 39.000
      }
   },
   "nvme-pci-0500":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 39.850,
         "temp1_max": 89.850,
         "temp1_min": -273.150,
         "temp1_crit": 94.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 39.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      },
      "Sensor 2":{
         "temp3_input": 42.850,
         "temp3_max": 65261.850,
         "temp3_min": -273.150
      }
   }
}
```

MusclePr, 2025-02-01:
Maybe ... this issue?
https://github.com/Meliox/PVE-mods/issues/65#issuecomment-2517164578

Other nodes screen shots.

[image]

[image]

Meliox, 2025-02-02:
Hello,

Yes, you're correct. It is related to having multiple nodes. There's no solution for that yet. I'll merge the issues.

Single node reference (ignore cpu type):
[image]

closed 2025-02-02 by Meliox