Meliox/PVE-mods #85: pve-mod-gui-sensors.sh not working in Proxmox 8.3.4

jauling, 2025-02-28:
**Describe the bug**
CPU temperatures not showing in Proxmox GUI. They did just fine in 8.3.3.

**To Reproduce**
Steps to reproduce the behavior:
```
# bash pve-mod-gui-sensors.sh install

Installing the Proxmox VE sensors display mod...

Preparing mod installation...

Do you wish to display temperatures for all cores [C] or just an average temperature per CPU [a] (note: AMD only supports average)? (C/a): 
[info] Temperatures will be displayed for all cores.

Detecting support for RAM temperature sensors...
[warning] No RAM temperature sensors found.

Detecting support for HDD/SDD temperature sensors...
[warning] No HDD/SSD temperature sensors found.

Detecting support for NVMe temperature sensors...
[warning] No NVMe temperature sensors found.

Detecting support for fan speed readings...
[warning] No fan speed sensors found.
type 1)
        Manufacturer: Default string
        Product Name: Default string
        Serial Number: Default string
type 2)
        Manufacturer: Default string
        Product Name: Default string
        Serial Number: Default string

Do you wish to enable system information? (1/2/n): 
[info] System information will be displayed...

Backup of "/usr/share/perl5/PVE/API2/Nodes.pm" saved to "/root/backup/Nodes.pm.2025-02-28_18-15-45".
Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/root/backup/pvemanagerlib.js.2025-02-28_18-15-45".
System information output added to "/usr/share/perl5/PVE/API2/Nodes.pm".
[warning] Sensor display items already added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".
```

[image]

```
# sensors
acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +32.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +30.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +30.0°C  (high = +105.0°C, crit = +105.0°C)
Core 2:        +30.0°C  (high = +105.0°C, crit = +105.0°C)
Core 3:        +30.0°C  (high = +105.0°C, crit = +105.0°C)
```

Meliox, 2025-02-28:
Can I ask for the sensors -j output instead?

Meliox, 2025-02-28:
Just tested. Still working in  8.3.4. You probably need a CTRL+F5 to refresh your browser.

jauling, 2025-03-01:
> Just tested. Still working in 8.3.4. You probably need a CTRL+F5 to refresh your browser.

Incognito browser also shows N/A, so I must be doing something wrong?

```
root@n100p4:~# bash pve-mod-gui-sensors.sh uninstall

Uninstalling the Proxmox VE sensors display mod...

Restoring modified files...
Copied latest backup to /usr/share/perl5/PVE/API2/Nodes.pm.
Copied latest backup to "/usr/share/pve-manager/js/pvemanagerlib.js".

Restarting PVE proxy...

root@n100p4:~# bash pve-mod-gui-sensors.sh install

Installing the Proxmox VE sensors display mod...

Preparing mod installation...

Do you wish to display temperatures for all cores [C] or just an average temperature per CPU [a] (note: AMD only supports average)? (C/a): 
[info] Temperatures will be displayed for all cores.

Detecting support for RAM temperature sensors...
[warning] No RAM temperature sensors found.

Detecting support for HDD/SDD temperature sensors...
[warning] No HDD/SSD temperature sensors found.

Detecting support for NVMe temperature sensors...
[warning] No NVMe temperature sensors found.

Detecting support for fan speed readings...
[warning] No fan speed sensors found.
type 1)
        Manufacturer: Default string
        Product Name: Default string
        Serial Number: Default string
type 2)
        Manufacturer: Default string
        Product Name: Default string
        Serial Number: Default string

Do you wish to enable system information? (1/2/n): 
[info] System information will be displayed...

Backup of "/usr/share/perl5/PVE/API2/Nodes.pm" saved to "/root/backup/Nodes.pm.2025-03-01_18-56-02".
Backup of "/usr/share/pve-manager/js/pvemanagerlib.js" saved to "/root/backup/pvemanagerlib.js.2025-03-01_18-56-02".
System information output added to "/usr/share/perl5/PVE/API2/Nodes.pm".
[warning] Sensor display items already added to the summary panel in "/usr/share/pve-manager/js/pvemanagerlib.js".

root@n100p4:~# sensors -j
{
   "acpitz-acpi-0":{
      "Adapter": "ACPI interface",
      "temp1":{
         "temp1_input": 27.800
      }
   },
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "Package id 0":{
         "temp1_input": 30.000,
         "temp1_max": 105.000,
         "temp1_crit": 105.000,
         "temp1_crit_alarm": 0.000
      },
      "Core 0":{
         "temp2_input": 30.000,
         "temp2_max": 105.000,
         "temp2_crit": 105.000,
         "temp2_crit_alarm": 0.000
      },
      "Core 1":{
         "temp3_input": 32.000,
         "temp3_max": 105.000,
         "temp3_crit": 105.000,
         "temp3_crit_alarm": 0.000
      },
      "Core 2":{
         "temp4_input": 32.000,
         "temp4_max": 105.000,
         "temp4_crit": 105.000,
         "temp4_crit_alarm": 0.000
      },
      "Core 3":{
         "temp5_input": 32.000,
         "temp5_max": 105.000,
         "temp5_crit": 105.000,
         "temp5_crit_alarm": 0.000
      }
   }
}
```

Gorgoroth-PL, 2025-03-02:
Mine also displays only N/A. I tested it on three completely different hardware configurations, although all have Intel processors and PVE 8.3.4

Meliox, 2025-03-02:
Thanks for the information. I think, I have enough to narrow down the issue for now.

Gorgoroth-PL, 2025-03-02:
If you want to test something, let me know 😉

Meliox, 2025-03-02:
I have found the error - let me think about the best solution :)

PR #86 (2025-03-02): Fix cpu detection
Regression from 3cb1ac8 as SENSORS_DETECTED is not true if sensors output only contains CPU. Resolves #85
closed 2025-03-02 by Meliox