Meliox/PVE-mods #89: drivetemp doesn't displayed USB attached SATA Devices

MyGH4me, 2025-03-18:
Is this a "Bug" or by Design?
The internal installed NVMe are OK

Drive(s)
NVMe Thermal State 
Drive 1: 36.9°C | Drive 2: 36.9°C

But the External SATA Drives with a SATA>USB3 Adapter are not display.

For Example:

These two SSD are attached with a single USB Adapter.
sda           8:0    0   1.9T  0 disk 
├─sda1        8:1    0   1.9T  0 part 
└─sda9        8:9    0     8M  0 part 
sdb           8:16   0   1.9T  0 disk 
├─sdb1        8:17   0   1.9T  0 part 
└─sdb9        8:25   0     8M  0 part

Meliox, 2025-03-18:
Please provide sensors -j output

MyGH4me, 2025-03-18:
Sorry missed it, my fault... Here is the JSON Output
root@pve:~# sensors -j

```
{
   "amdgpu-pci-0b00":{
      "Adapter": "PCI adapter",
      "vddgfx":{
ERROR: Can't get value of subfeature in0_input: Can't read

      },
      "vddnb":{
ERROR: Can't get value of subfeature in1_input: Can't read

      },
      "edge":{
         "temp1_input": 39.000
      }
   },
   "ucsi_source_psy_USBC000:001-isa-0000":{
      "Adapter": "ISA adapter",
      "in0":{
         "in0_input": 0.000,
         "in0_min": 0.000,
         "in0_max": 0.000
      },
      "curr1":{
         "curr1_input": 0.000,
         "curr1_max": 0.000
      }
   },
   "nvme-pci-0800":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 36.850,
         "temp1_max": 99.850,
         "temp1_min": -0.150,
         "temp1_crit": 109.850,
         "temp1_alarm": 0.000
      }
   },
   "acpitz-acpi-0":{
      "Adapter": "ACPI interface",
      "temp1":{
         "temp1_input": 30.000
      }
   },
   "k10temp-pci-00c3":{
      "Adapter": "PCI adapter",
      "Tctl":{
         "temp1_input": 39.750
      }
   },
   "nvme-pci-0a00":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 35.850,
         "temp1_max": 99.850,
         "temp1_min": -0.150,
         "temp1_crit": 109.850,
         "temp1_alarm": 0.000
      }
   }
}
```

Meliox, 2025-03-18:
As the information is not available here, there is no way to show it. If you can figure out a way to get the data displayed with sensors, then we can find a way. Sorry.

MyGH4me, 2025-03-18:
SMART gets some Informations, but depending on the Manufactur. Samsung SMART is "190 Airflow_Temperature_Cel" and Transcend with "190 Airflow_Temperature_Cel"

Meliox, 2025-03-18:
I am not really into pulling data from smart. It is slow (couple of seconds), requires workarounds to mitigate, and would change the whole implementation. So unless a you find a way to get til into sensors, then I cannot implement it.

closed 2025-03-18 by MyGH4me