Meliox/PVE-mods #91: HDD names

fjk8, 2025-03-21:
Can you add option to have HDD names instead of Drive 1,2,3? 
f.e. INTEL_SSDSCKKF512G8 instead of Drive 1

Meliox, 2025-03-22:
I like the idea.

Could you try to add a label according to 5.1.2 here https://wiki.archlinux.org/title/Lm_sensors
And the provide sensors -j output. Though I am unsure if it is working; https://github.com/lm-sensors/lm-sensors/pull/371 https://github.com/lm-sensors/lm-sensors/issues/230. The project seems to be abandoned at this stage, so.... I'd rather not make an extensive workaround.

fjk8, 2025-03-23:
As you write, I'm unable to label drives because "Undeclared bus id referenced" error :(

fjk8, 2025-03-25:
Maybe you can add drive or device names from somewhere else than sensors?

vit9696, 2025-05-03:
@Meliox, perhaps you could consider reading temperature data directly from `drivetemp` module rather than try to parse abandoned lm-sensors output? What we need are `temp1_input` and `temp1_crit` from `sysfs`, see the [docs](https://docs.kernel.org/hwmon/drivetemp.html).

On my system I can find the devices using the following `find` command:

```
# find /sys/devices/ -path '*ata*' -name temp1_input
/sys/devices/pci0000:00/0000:00:1f.2/ata1/host0/target0:0:0/0:0:0:0/hwmon/hwmon3/temp1_input
/sys/devices/pci0000:00/0000:00:1f.2/ata4/host3/target3:0:0/3:0:0:0/hwmon/hwmon6/temp1_input
/sys/devices/pci0000:00/0000:00:1f.2/ata5/host4/target4:0:0/4:0:0:0/hwmon/hwmon7/temp1_input
/sys/devices/pci0000:00/0000:00:1f.2/ata3/host2/target2:0:0/2:0:0:0/hwmon/hwmon5/temp1_input
```

Personally I do not quite need names, but rather correct indices in the bay, so for me something like this works best:

```
# find /sys/devices/ -path '*ata*' -name temp1_input -exec bash -c 'b=$(dirname "$0"); a=$(echo "$0" | grep -oE "ata[0-9]+"); cur=$(($(cat "$b/temp1_input") / 1000)); if [ -f "$b/temp1_crit" ]; then crit=$(($(cat "$b/temp1_crit") / 1000)); else crit=0 ; fi ; echo "$a $cur / $crit"' {} \;
ata1 38 / 55
ata4 26 / 0
ata5 30 / 0
ata3 35 / 55
```

However, one can always get names from `sysfs` e.g. from `model` file.

As a side note, I should mention that one of my SSD drives reports 0 critical minimal and maximum temperature, so both missing file and file containing 0 value need to be handled:

```
# cat /sys/devices/pci0000:00/0000:00:1f.2/ata4/host3/target3:0:0/3:0:0:0/hwmon/hwmon6/temp1_crit
0
# cat /sys/devices/pci0000:00/0000:00:1f.2/ata4/host3/target3:0:0/3:0:0:0/hwmon/hwmon6/temp1_lcrit
0
```

In fact, even with lm-sensors right now I get constantly red temperature for this drive due to any current value being always bigger than 0 critical value.

vit9696, 2025-05-03:
To think of it, given the request matter, users who need it likely want to have more control on sensor filtering. Perhaps the right approach here is to allow specifying a _custom command to get sensor output_ and define the expected format? This way one can replace `sensors -j` output with something custom, correcting/adding/removing/renaming values as needed.

Right now, it is already semi-possible to customise things by creating a script in `/usr/sbin/sensors` and teaching it to handle the `-j` argument. I made one to produce fixed drive count in the right order with sane values regardless of whether the drives are present in the bay. Here is what I got:

```python
#!/usr/bin/env python3

import json
import sys
import subprocess

SENSORS_UTIL = '/usr/bin/sensors'
ALLOWED_SENSORS = {
    'drivetemp-scsi-0-0': {'Adapter': 'SCSI adapter'},
    'drivetemp-scsi-1-0': {'Adapter': 'SCSI adapter'},
    'drivetemp-scsi-2-0': {'Adapter': 'SCSI adapter'},
    'drivetemp-scsi-3-0': {'Adapter': 'SCSI adapter'},
    'drivetemp-scsi-4-0': {'Adapter': 'SCSI adapter'},
    'coretemp-isa-0000': {'Adapter': 'ISA adapter'},
    'nct6791-isa-0290': {'Adapter': 'ISA adapter', 'AllowedKeys': ['fan2']},
}

def hook_sensors():
    argv = [SENSORS_UTIL, '-j']
    res = subprocess.run(argv, stdin=sys.stdin, check=False, capture_output=True)
    if res.returncode != 0:
       print("ERROR: Failed to call sensors")
       return res.returncode

    sensor_data = json.loads(res.stdout.decode('UTF-8'))
    filtered_data = {}

    for sensor, request in ALLOWED_SENSORS.items():
        data = sensor_data.get(sensor)
        if not data:
            data = {'Adapter': request['Adapter']}
        if request['Adapter'] == 'SCSI adapter':
            if 'temp1' not in data:
                data['temp1'] = {}
            if 'temp1_input' not in data['temp1']:
                data['temp1']['temp1_input'] = 0.0
            if data['temp1'].get('temp1_max') == 0.0:
                del data['temp1']['temp1_max']
            if data['temp1'].get('temp1_crit') == 0.0:
                del data['temp1']['temp1_crit']
        allowed = request.get('AllowedKeys')
        if allowed:
            data = {k: v for k, v in data.items() if k in allowed or k == 'Adapter'}
        filtered_data[sensor] = data

    print(json.dumps(filtered_data, indent=2))
    return 0

def main(argv):
    if len(argv) == 2 and argv[1] == '-j':
       return hook_sensors()
    argv[0] = SENSORS_UTIL
    return subprocess.run(argv, stdin=sys.stdin, stdout=sys.stdout, stderr=sys.stderr, check=False).returncode

if __name__ == "__main__":
    sys.exit(main(sys.argv))
```

It remains to be semi-possible in regards of the original issue, because the drive names are fixed as `Drive 1`, `Drive 2`, etc. Perhaps the format could define an optional 'FriendlyName' field, which others could fill to specify drive names they want to see instead of the generic `Drive #`.