Meliox/PVE-mods #98: Summary page updates irregularly

Strux-DK, 2025-04-06:
**Describe the bug**
A clear and concise description of what the bug is.

Before installing your script, i saw the summary page updating persistently once every second. Now it updates, on average, once every 1-3 seconds. Do you have any idea why that is or what can be done to improve the update time?

Also, is it possible to save the temps over time, in a file? I would like this because my SFF PC running proxmox is running hot, and i would like a baseline to work from, to verify which of my solutions work. That's actually what made me find your script, lol.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to your github
2. run three commands
3. force update browser
4. See irregular updates of proxmox summary screen

**Expected behavior**
Persistent updates of the proxmox summary screen every 1 seconds

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Sensors output**
If applicable, provide sensor datadump (use the argument save-sensors-data)

**Additional context**
Add any other context about the problem here.

Meliox, 2025-04-06:
On every refresh it pulls data from sensor -j. Should there be any delay running this command, then that will also be reflected in the webui. I have not that large a delay before, but there has been reports about increased i/o. How long time does yours take to run? Are there any errors or slow reads from specific sensors. 
If you want to play with a workaround, you could try below

while true; do
    sensors -j >> output.json
    echo "," >> output.json
    sleep 1
done
 
and then change thr script to debug and change the file location to the one used above. 

I can only think about storing sensor data to a file/memory and then read it from there instead. That would require a service to be running in the background though. Kinda the way I attempted to implement graphic cards load and temp. That increased the i/o/cpu usage a fair bit though. 

I think you would be looking for dashboards like grafana for analysis and then a database to store the data in. I don't have any recommendations here.

Strux-DK, 2025-04-06:
>How long time does yours take to run? Are there any errors or slow reads from specific sensors. 

How can i check this? I'm not sure how to verify this, sorry

>If you want to play with a workaround, you could try below
> 
> while true; do sensors -j > output.json sleep 1 done
> 
> and then change thr script to debug and change the file location to the one used above.

I'm far from a developer and have minimal experience with scripting. I'm not sure what this does or where to put it in the script.

> I can only think about storing sensor data to a file/memory and then read it from there instead. That would require a service to be running in the background though. Kinda the way I attempted to implement graphic cards load and temp. That increased the i/o/cpu usage a fair bit though.

I was just thinking, with my limited knowledge on this, that your script must run some command, that outputs the temps and then it modifies the summary page and shows the temps there, right?
So would it be possible to take the output from the command, maybe modify it a bit to show something like the following and then saving it in a .txt file? I'll use something like excel to work with the data, i just need temps over time.
Desired output to textfile: 
Date, time
sensor 1 name: sensor 1 temp
sensor 2 name: sensor 2 temp
etc.
Example: 06-04-2025 14:20:05
CPU temp: 76 C
NVME temp: 64 C

> I think you would be looking for dashboards like grafana for analysis and then a database to store the data in. I don't have any recommendations here.

My immediate needs are just for a couple of days. However, it would be nice if it was possible to make graphs of the temps over time, in the summary page, like with CPU and RAM usage.