Meliox/PVE-mods #99: Choose between sensors for other HW, like the CPU

Strux-DK, 2025-04-08:
**Is your feature request related to a problem? Please describe.**
My NVME has 2 sensors. I don't know why, but one of the sensors (Sensor 1) is constantly at 43.9C. Th same value is used for the comp. temp. sensor. This is the sensor your script chooses to show. I just reran the script and got a question on whether i wanted temps for all CPU cores or just an average, but got no questions about the rest of the temp sensors. 

**Describe the solution you'd like**
Would this be possible, so i eventually can choose the other sensor (Sensor 2), that actually fluctuates?

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

**Sensors output**
root@pve:~# sensors
nvme-pci-0100
Adapter: PCI adapter
Composite:    +43.9°C  (low  = -273.1°C, high = +67.8°C)
                       (crit = +70.8°C)
Sensor 1:     +43.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +47.9°C  (low  = -273.1°C, high = +65261.8°C)

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +55.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +55.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:        +51.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:        +56.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:        +48.0°C  (high = +80.0°C, crit = +100.0°C)

root@pve:~#

Meliox, 2025-04-08:
It's actually using Composite: +43.9°C (low = -273.1°C, high = +67.8°C).

Just tested on my system and it did increase under workload.

 Can you reconfirm the behavior?

If you need to apply load it can be done with below script. Save it in load test.sh and then bash test.sh.

```bash
#!/bin/bash

for i in {1..20}; do
    echo "Creating 1GB file (iteration $i)..."
    dd if=/dev/zero of=dummyfile.bin bs=1M count=4196 status=none
    echo "Deleting dummyfile.bin (iteration $i)..."
    rm -f dummyfile.bin
done
```

That said, it can be made configurable.. It is just if you have several nvme - then maybe it should be selected for all. Hmm

AnAnalogGuy, 2025-05-29:
This said, looking at 

sensors
**spd5118-i2c-0-50**
Adapter: SMBus I801 adapter at efa0
**ERROR: Can't get value of subfeature temp1_max_alarm: Can't read**
temp1:        **+46.2°C**  (low  =  +0.0°C, high = +55.0°C)
                       (crit low =  +0.0°C, crit = +85.0°C)

the script is not able to detect this RAM temp sensor, although it's there and provides correct values. 
RAM is a Crucial DDR5 4800 MHZ 32 GB Module. 
Is there anything I can do to make the script detect this sensor?

Jiinja, 2025-11-02:
@AnAnalogGuy I got the script to detect my DDR5 ram sensors by adding an override file for the output of sensors. 

nano /etc/sensors.d/ram.conf
`chip "spd5118-*"
    label temp1 "SODIMM"`

All of my DDR5 sticks start with spd5118- in the sensors output, but show temps as temp1. The script is looking for SODIMM where temp1 is, so this replaces temp1 with SODIMM. With this override, the output on my machine looks like this 

[image]

Additionally I made a few other changes in a similar config file to label my fans as chassis, cpu, hba, and gpu instead of just fan1 1, fan2 2, fan3 3, etc.

jimninetoes, 2025-12-17:
@Jiinja i have tried using your script in my machine however i keep getting and error "Error: File /etc/sensors.d/ram.conf, line 1: Parse error in chip name"

i changed the name to the corresponding name given with "sensors" output