Meliox/PVE-mods #100: on MS-01 CPU temp is not displayed

stanthewizzard, 2025-04-14:
Hello

With sensors I get this

```
nct6798-isa-0a20
Adapter: ISA adapter
fan1:         2331 RPM  (min =    0 RPM)
fan2:         1869 RPM  (min =    0 RPM)
vrm:           +45.5°C  (high = +80.0°C, hyst = +75.0°C)
                        (crit = +95.0°C)  sensor = thermistor
CPU:           +48.0°C  (high = +80.0°C, hyst = +75.0°C)
PCH_CPU_TEMP:   +0.0°C
PCH_MCH_TEMP:   +0.0°C
beep_enable:  enabled

nvme-pci-5800
Adapter: PCI adapter
Composite:    +41.9°C  (low  =  -0.1°C, high = +82.8°C)
                       (crit = +84.8°C)
Sensor 1:     +41.9°C  (low  = -273.1°C, high = +65261.8°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C

coretemp-isa-0000
Adapter: ISA adapter
6P 8E package:  +45.0°C  (high = +100.0°C, crit = +100.0°C)
P Core 1:       +42.0°C  (high = +100.0°C, crit = +100.0°C)
P Core 2:       +44.0°C  (high = +100.0°C, crit = +100.0°C)
P Core 3:       +45.0°C  (high = +100.0°C, crit = +100.0°C)
P Core 4:       +42.0°C  (high = +100.0°C, crit = +100.0°C)
P Core 5:       +44.0°C  (high = +100.0°C, crit = +100.0°C)
P Core 6:       +41.0°C  (high = +100.0°C, crit = +100.0°C)
E Core 1:       +43.0°C  (high = +100.0°C, crit = +100.0°C)
E Core 2:       +43.0°C  (high = +100.0°C, crit = +100.0°C)
E Core 3:       +43.0°C  (high = +100.0°C, crit = +100.0°C)
E Core 4:       +43.0°C  (high = +100.0°C, crit = +100.0°C)
E Core 5:       +40.0°C  (high = +100.0°C, crit = +100.0°C)
E Core 6:       +41.0°C  (high = +100.0°C, crit = +100.0°C)
E Core 7:       +41.0°C  (high = +100.0°C, crit = +100.0°C)
E Core 8:       +41.0°C  (high = +100.0°C, crit = +100.0°C)

nvme-pci-0100
Adapter: PCI adapter
Composite:    +41.9°C  (low  = -273.1°C, high = +89.8°C)
                       (crit = +94.8°C)
Sensor 1:     +41.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +47.9°C  (low  = -273.1°C, high = +65261.8°C)
```

But with the script I get this

```
CPU(s) 20 x 12th Gen Intel(R) Core(TM) i9-12900H (1 Socket)
--
CPU Thermal State N/A
Cooling
Fan Speed(s) Fan1: 2323 RPM \| Fan2: 1904 RPM
Drive(s)
NVMe Thermal State Drive 1: 41.9°C \| Drive 2: 41.9°C
 
Kernel Version Linux 6.8.12-9-pve (2025-03-16T19:18Z)

```

What do I need to do to have the temps ?
Thanks

Meliox, 2025-04-20:
Please provide sensor -j output. I think the temps of your cores are listed different, hence a change will be needed.

stanthewizzard, 2025-04-21:
sensors -j

```
{
   "nct6798-isa-0a20":{
      "Adapter": "ISA adapter",
      "fan1":{
         "fan1_input": 2319.000,
         "fan1_min": 0.000,
         "fan1_alarm": 0.000,
         "fan1_beep": 0.000,
         "fan1_pulses": 2.000
      },
      "fan2":{
         "fan2_input": 1914.000,
         "fan2_min": 0.000,
         "fan2_alarm": 0.000,
         "fan2_beep": 0.000,
         "fan2_pulses": 2.000
      },
      "vrm":{
         "temp2_input": 45.500,
         "temp2_max": 80.000,
         "temp2_max_hyst": 75.000,
         "temp2_crit": 95.000,
         "temp2_alarm": 0.000,
         "temp2_type": 4.000,
         "temp2_offset": 0.000,
         "temp2_beep": 0.000
      },
      "CPU":{
         "temp7_input": 49.000,
         "temp7_max": 80.000,
         "temp7_max_hyst": 75.000,
         "temp7_alarm": 0.000,
         "temp7_beep": 0.000
      },
      "PCH_CPU_TEMP":{
         "temp11_input": 0.000
      },
      "PCH_MCH_TEMP":{
         "temp12_input": 0.000
      },
      "beep_enable":{
         "beep_enable": 1.000
      }
   },
   "nvme-pci-5800":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 40.850,
         "temp1_max": 82.850,
         "temp1_min": -0.150,
         "temp1_crit": 84.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 40.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      }
   },
   "acpitz-acpi-0":{
      "Adapter": "ACPI interface",
      "temp1":{
         "temp1_input": 27.800
      }
   },
   "coretemp-isa-0000":{
      "Adapter": "ISA adapter",
      "6P 8E package":{
         "temp1_input": 45.000,
         "temp1_max": 100.000,
         "temp1_crit": 100.000,
         "temp1_crit_alarm": 0.000
      },
      "P Core 1":{
         "temp2_input": 44.000,
         "temp2_max": 100.000,
         "temp2_crit": 100.000,
         "temp2_crit_alarm": 0.000
      },
      "P Core 2":{
         "temp6_input": 43.000,
         "temp6_max": 100.000,
         "temp6_crit": 100.000,
         "temp6_crit_alarm": 0.000
      },
      "P Core 3":{
         "temp10_input": 47.000,
         "temp10_max": 100.000,
         "temp10_crit": 100.000,
         "temp10_crit_alarm": 0.000
      },
      "P Core 4":{
         "temp14_input": 41.000,
         "temp14_max": 100.000,
         "temp14_crit": 100.000,
         "temp14_crit_alarm": 0.000
      },
      "P Core 5":{
         "temp18_input": 43.000,
         "temp18_max": 100.000,
         "temp18_crit": 100.000,
         "temp18_crit_alarm": 0.000
      },
      "P Core 6":{
         "temp22_input": 43.000,
         "temp22_max": 100.000,
         "temp22_crit": 100.000,
         "temp22_crit_alarm": 0.000
      },
      "E Core 1":{
         "temp26_input": 44.000,
         "temp26_max": 100.000,
         "temp26_crit": 100.000,
         "temp26_crit_alarm": 0.000
      },
      "E Core 2":{
         "temp27_input": 44.000,
         "temp27_max": 100.000,
         "temp27_crit": 100.000,
         "temp27_crit_alarm": 0.000
      },
      "E Core 3":{
         "temp28_input": 44.000,
         "temp28_max": 100.000,
         "temp28_crit": 100.000,
         "temp28_crit_alarm": 0.000
      },
      "E Core 4":{
         "temp29_input": 44.000,
         "temp29_max": 100.000,
         "temp29_crit": 100.000,
         "temp29_crit_alarm": 0.000
      },
      "E Core 5":{
         "temp30_input": 40.000,
         "temp30_max": 100.000,
         "temp30_crit": 100.000,
         "temp30_crit_alarm": 0.000
      },
      "E Core 6":{
         "temp31_input": 40.000,
         "temp31_max": 100.000,
         "temp31_crit": 100.000,
         "temp31_crit_alarm": 0.000
      },
      "E Core 7":{
         "temp32_input": 40.000,
         "temp32_max": 100.000,
         "temp32_crit": 100.000,
         "temp32_crit_alarm": 0.000
      },
      "E Core 8":{
         "temp33_input": 41.000,
         "temp33_max": 100.000,
         "temp33_crit": 100.000,
         "temp33_crit_alarm": 0.000
      }
   },
   "nvme-pci-0100":{
      "Adapter": "PCI adapter",
      "Composite":{
         "temp1_input": 41.850,
         "temp1_max": 89.850,
         "temp1_min": -273.150,
         "temp1_crit": 94.850,
         "temp1_alarm": 0.000
      },
      "Sensor 1":{
         "temp2_input": 41.850,
         "temp2_max": 65261.850,
         "temp2_min": -273.150
      },
      "Sensor 2":{
         "temp3_input": 45.850,
         "temp3_max": 65261.850,
         "temp3_min": -273.150
      }
   }
}
```

Thanks