Meliox/PVE-mods #102: Raspberry PI support

kheops31, 2025-05-31:
Please consider Raspberry PI support.
I own RPI 3/4 if you need extra information.

Regards.

Meliox, 2025-05-31:
You have not provided any documentation of the sorts, sensor output, what the issue is etc.

closed 2025-05-31 by Meliox

kheops31, 2025-05-31:
You totally right,

Issue : The Proxmox dashboard after the script install, doesn't output cpu temperature.

**System** 
RPI 4 
Debian 12 
Proxmox 8,3.3
(Linux PiMox 6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux)

**sensor command output**
cpu_thermal-virtual-0
Adapter: Virtual device
temp1:        +56.5°C

rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A

I've read your the script and seems that you look for word "core" as search criteria, and It doesn't exist on "sensor"  output.

Hope that information are enough.

Regards.

Vaskyy, 2025-11-21:
Since hes not answering, ima put my output here:
```
vaskyy@Vaskyy-Five:~ $ sudo sensors -j 2>/dev/null | python3 -m json.tool
{
    "hidpp_battery_6-hid-3-3b": {
        "Adapter": "HID adapter",
        "in0": {
            "in0_input": 3.803
        }
    },
    "rpi_volt-isa-0000": {
        "Adapter": "ISA adapter",
        "in0": {
            "in0_lcrit_alarm": 0.0
        }
    },
    "nvme-pci-10600": {
        "Adapter": "PCI adapter",
        "Composite": {
            "temp1_input": 32.85,
            "temp1_max": 76.85,
            "temp1_min": -273.15,
            "temp1_crit": 78.85,
            "temp1_alarm": 0.0
        },
        "Sensor 1": {
            "temp2_input": 32.85,
            "temp2_max": 65261.85,
            "temp2_min": -273.15
        },
        "Sensor 2": {
            "temp3_input": 34.85,
            "temp3_max": 65261.85,
            "temp3_min": -273.15
        }
    },
    "cpu_thermal-virtual-0": {
        "Adapter": "Virtual device",
        "temp1": {
            "temp1_input": 45.2
        }
    },
    "rp1_adc-isa-0000": {
        "Adapter": "ISA adapter",
        "in1": {
            "in1_input": 1.463
        },
        "in2": {
            "in2_input": 2.465
        },
        "in3": {
            "in3_input": 1.216
        },
        "in4": {
            "in4_input": 1.159
        },
        "temp1": {
            "temp1_input": 51.985
        }
    },
    "pwmfan-isa-0000": {
        "Adapter": "ISA adapter",
        "fan1": {
            "fan1_input": 0.0
        }
    },
    "nvme-pci-10400": {
        "Adapter": "PCI adapter",
        "Composite": {
            "temp1_input": 29.85,
            "temp1_max": 84.85,
            "temp1_min": -0.15,
            "temp1_crit": 94.85,
            "temp1_alarm": 0.0
        },
        "Sensor 1": {
            "temp2_input": 29.85,
            "temp2_max": 65261.85,
            "temp2_min": -273.15
        },
        "Sensor 2": {
            "temp3_input": 31.85,
            "temp3_max": 65261.85,
            "temp3_min": -273.15
        },
        "Sensor 8": {
            "temp9_input": 29.85,
            "temp9_max": 65261.85,
            "temp9_min": -273.15
        }
    }
}
```

Im using a Raspberry Pi 5 with Lierfang Pxvirt 8.4.10 installed:
`Linux Vaskyy-Five 6.12.47+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.47-1+rpt1~bookworm (2025-09-16) aarch64 GNU/Linux`

CPU (cpu_thermal-virtual-0) and RPI1 I/O Controller (rp1_adc-isa-0000) Temps are not being recognized by the mod script.

Everything else did get displayed in the UI perfectly.

Vaskyy, 2025-11-22:
Is that information useful in any case @Meliox ?

Meliox, 2026-01-04:
@Vaskyy: Do you have an update sensors output given the referred update?