Meliox/PVE-mods #105: P.V.E. 9.0.3

jinsei8, 2025-08-06:
Thermal States for CPU and NVME are reporting as N/A after updating from 8.4.8 to 9.0.3. It seems to not survive any version upgrades 8.4.5 -> 8.4.6 I had to reinstall. Following normal install instructions results in N/A. Thanks.

closed 2025-08-06 by Meliox

jinsei8, 2025-08-07:
Ran through normal install procedure. Still:

CPU(s)
CPU Thermal State - N/A
Drive(s)
NVMe Thermal State - N/A

Will uninstall and wait for your next push. Thanks 👍

dorsanty, 2025-08-07:
I ran wget for the latest pve-mod-nag-screen.sh (as per the [install](https://github.com/Meliox/PVE-mods#install) doc) and it resolved the problem for me. I get CPU and Drive thermal data.

If you re-downloaded the script into the same location as the old one did wget write it as pve-mod-nag-screen.sh.1 ? Possibly resulting in re-running the older script, not the newest one.

Meliox, 2025-08-07:
@jinsei8 I believe, I have found the culprit. As I don't have your sensors output I cannot confirm it, but at least it fixes the reported error in #108.

@dorsanty: That's true, but I do not think that's the reason here though.

PR #106 (2025-08-06): Fix sensors JSON output to formatted version
fixes #105