Meliox/PVE-mods #114: cant' uninstall/install to update

dtrywets, 2025-08-12:
Hi, 
I was trying to uninstall and reinstall to perform such thing as "update".
But sadly neither worked at my installation:

`root@pxmx:~# cd scripts
root@pxmx:~/scripts# bash pve-mod-gui-sensors.sh uninstall

Uninstalling the Proxmox VE sensors display mod...

Restoring modified files...
[warning] No Nodes.pm files found.
[warning] No pvemanagerlib.js files found.

root@pxmx:~/scripts# bash <(curl -fsSL https://raw.githubusercontent.com/Meliox/PVE-mods/refs/heads/main/pve-mod-gui-sensors.sh) uninstall

Uninstalling the Proxmox VE sensors display mod...

Restoring modified files...
[warning] No Nodes.pm files found.
[warning] No pvemanagerlib.js files found.

root@pxmx:~/scripts# bash <(curl -fsSL https://raw.githubusercontent.com/Meliox/PVE-mods/refs/heads/main/pve-mod-gui-sensors.sh) install

Installing the Proxmox VE sensors display mod...
[error] Mod is already installed. Uninstall existing before installing.
root@pxmx:~/scripts#`

is there some sort of workaround for that case?

Meliox, 2025-08-13:
And sorry.. The Workaround is to remove the exit here. https://github.com/Meliox/PVE-mods/blob/e8af05c77029e4eaba524d5dd25977f55ac21f53/pve-mod-gui-sensors.sh#L280

devdecrux, 2025-09-16:
I have the same problem as the colleague who opened this issue. When I try to install:
`root@pve1:/opt# bash pve-mod-gui-sensors.sh install

Installing the Proxmox VE sensors display mod...

=== Preparing mod installation ===
[info] Root privileges verified.
[error] Mod is already installed. Uninstall existing before installing.`

No backup files are found:
`root@pve1:/opt# find / -name "Nodes.pm"
/usr/share/perl5/PVE/API2/Nodes.pm
root@pve1:/opt# find / -name "pvemanagerlib.js"
/usr/share/pve-manager/js/pvemanagerlib.js`

and when I uninstall with the script:
`root@pve1:/opt# bash pve-mod-gui-sensors.sh uninstall

Uninstalling the Proxmox VE sensors display mod...
=== Uninstalling Mod ===
[info] Root privileges verified.
[info] Using default backup directory: /root/PVE-MODS
[info] Restoring modified files...
[warning] No Nodes.pm backup files found.
[warning] No pvemanagerlib.js backup files found.

Clear the browser cache to ensure all changes are visualized. (any key to continue):`

Meliox, 2025-09-17:
@devdecrux please see referred post