Meliox/PVE-mods #138: UPS output shows N/A

OiCkilL, 2025-09-15:
**Describe the bug**
UPS output shows N/A

**Screenshots**

[image]

[image]

**Sensors output**
# upsc ups@localhost
Init SSL without certificate database
battery.capacity: 100
battery.charge: 98
battery.charge.low: 25
battery.charge.warning: 5
battery.current: 0.000
battery.mfr.date: OPEN
battery.runtime: 2834
battery.temperature: 30
battery.type: LiFePO4
battery.voltage: 13.50
device.mfr: WL
device.model: W120
device.serial: 123456788888
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.name: usbhid-ups
driver.parameter.override.battery.charge.low: 25
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1
driver.version.data: openUPS HID 0.5
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.28 (API: 0x100010a)
input.current: 6.486
input.voltage: 12.01
output.current: 6.486
output.voltage: 12.01
ups.alarm: No battery installed!
ups.mfr: WL
ups.model: W120
ups.productid: d005
ups.serial: 123456788888
ups.status: ALARM OL
ups.vendorid: 04d8
**Additional context**
Add any other context about the problem here.

pear235, 2025-09-17:
I'm also getting the N/A for UPS, yet i seem to have all the required values :

root@pve:~# upsc pxups@localhost:3493
Init SSL without certificate database
battery.charge: 92
battery.charge.low: 15
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 2438
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CP1500EPFCLCD
device.model: CRJB103.551
device.serial: CPS
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.flag.ignorelb: enabled
driver.name: usbhid-ups
driver.parameter.override.battery.charge.low: 15
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 10
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1
driver.version.data: CyberPower HID 0.8
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.28 (API: 0x100010a)
input.transfer.high: 265
input.transfer.low: 175
input.voltage: 236.0
input.voltage.nominal: 230
output.voltage: 236.0
ups.beeper.status: disabled
ups.delay.shutdown: 20
ups.delay.start: 30
**ups.load: 16**
ups.mfr: CP1500EPFCLCD
ups.model: CRJB103.551
ups.productid: 0501
**ups.realpower.nominal: 900**
ups.serial: CPS
ups.status: OL CHRG
**ups.test.result: Done and passed**
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
root@pve:~#

pear235, 2025-09-17:
I did do the ctrl+r, yea.

Not sure if this is the screenshot you wanted, if not, pls let me know :

[image]

Meliox, 2026-08-17:
Is there still any open issues here?

OiCkilL, 2026-08-18:
> Is there still any open issues here?

Yes, it’s still unresolved. I could ask an AI to help identify the issue, but I’d appreciate it if you could provide some prompt suggestions to assist me in debugging.

Meliox, 2026-08-18:
Alright. Can I ask you to upgrade to the new version and provide a screenshot as well as providing the ups output. Thanks

OiCkilL, 2026-08-20:
Thanks for the follow-up, and sorry for the delay.

There were **two separate issues** mixed in this thread.

### 1. `ups.alarm: No battery installed!` / `ALARM` in `ups.status`

This was **not a PVE-mods bug**, and it was also not because the W120 has a sealed/non-replaceable battery.

The W120 is claimed by NUT's `openUPS HID` subdriver (`04d8:d005`, `driver.version.data: openUPS HID 0.5`). On NUT **2.8.1** (current Debian / Proxmox package), `openups-hid.c` maps HID `BatteryPresent` with **inverted polarity**:

```c
static const char *openups_nobattery_fun(double value)
{
    return value ? "nobattery" : "!nobattery";
}
```

So `BatteryPresent=1` (battery is present) became the `nobattery` alarm. Charge/runtime/voltage were already valid; only that flag was wrong.

Upstream already documented the fix: *openups-hid had nobattery definition inverted, causing alarms* (NUT NEWS; same symptom on the related W150 in [networkupstools/nut#3246](https://github.com/networkupstools/nut/issues/3246)). Newer NUT uses the shared `nobattery_info` table instead.

I rebuilt `usbhid-ups` from 2.8.1 with that polarity corrected and replaced the driver on this host (`dpkg-divert` so package upgrades keep the Debian binary aside). After restart:

```
# upsc ups@localhost
Init SSL without certificate database
battery.capacity: 100
battery.charge: 98
battery.charge.low: 10
battery.charge.warning: 20
battery.current: 0.080
battery.mfr.date: OPEN
battery.runtime: 3108
battery.temperature: 30
battery.type: LiFePO4
battery.voltage: 13.50
device.mfr: WL
device.model: W120
device.serial: 123456788888
device.type: ups
driver.debug: 0
driver.flag.allow_killpower: 0
driver.flag.onlinedischarge_calibration: enabled
driver.name: usbhid-ups
driver.parameter.override.battery.charge.low: 10
driver.parameter.override.battery.charge.warning: 20
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: quiet
driver.version: 2.8.1
driver.version.data: openUPS HID 0.5
driver.version.internal: 0.52
driver.version.usb: libusb-1.0.28 (API: 0x100010a)
input.current: 3.813
input.voltage: 11.89
output.current: 3.733
output.voltage: 11.89
ups.mfr: WL
ups.model: W120
ups.productid: d005
ups.serial: 123456788888
ups.status: OL CHRG
ups.vendorid: 04d8
```

`ups.alarm` is gone, `ups.status` is `OL` / `OL CHRG` with no `ALARM`. Confirmed after a full driver poll cycle.

`battery.mfr.date: OPEN` is still not a manufacture date — the subdriver reads firmware `iOEMInformation`, and openUPS reports the string `OPEN`. Overriding it in `ups.conf` is fine if you want a reminder date.

### 2. Load / Output / Test still N/A

That part is still as you described earlier: this DC UPS does **not** publish `ups.load`, `ups.realpower.nominal`, or `ups.test.result`. The widget currently needs those keys, so N/A is expected.

For this model, Output can be derived as `output.current * output.voltage` (about 3.7 A × 11.9 V ≈ 44 W here). Load could be that wattage / a configured nominal. There is no self-test result from the device.

Happy to retest a fallback in the widget if you add one (e.g. Output from `output.current * output.voltage` when `ups.load` / `ups.realpower.nominal` are missing).

Meliox, 2026-08-20:
Thank you.

I will have look at a fall back for calculating the output.

I think it should be OK to leave ```ups.test.result``` showing N/A.

```battery.mfr.date ``` a test could be made to check if this is a date otherwise leave it as N/A.

Meliox, 2026-08-21:
Would you be able to test referred PR?

OiCkilL, 2026-08-21:
Tested #309 on this W120.

Output is no longer blocked: collector now sets `ups.realpower` from `output.current * output.voltage` (e.g. 3.531 A × 11.91 V → 42 W). `battery.mfr.date: OPEN` is stripped, so MFD shows N/A. Load and Test stay N/A because the device still has no `ups.load` / `ups.test.result`.

Details on the PR: https://github.com/Meliox/PVE-mods/pull/309

OiCkilL, 2026-08-21:
Opened a follow-up PR with the compact UPS row we tested on this W120, plus hiding empty GPU Details/Fans: https://github.com/Meliox/PVE-mods/pull/310